takaswie/linux-firewire-dkms #11: Error using DKMS on Manjaro linux to backport snd-firewire-improve

quietpinguin, 2015-01-28:
Hi Takashi

Can you help? I would like to try out the snd_dice driver with a Focusrite Saffire PRO 24.

Here is the make log...

DKMS make.log for alsa-firewire-3.11 for kernel 3.18.3-1-MANJARO (x86_64)
Mi 28. Jan 20:48:54 CET 2015
make -C /lib/modules/3.18.3-1-MANJARO/build M=/var/lib/dkms/alsa-firewire/3.11/build modules
make[1]: Entering directory '/usr/lib/modules/3.18.3-1-MANJARO/build'
  CC [M]  /var/lib/dkms/alsa-firewire/3.11/build/sound/firewire/lib.o
  CC [M]  /var/lib/dkms/alsa-firewire/3.11/build/sound/firewire/iso-resources.o
  CC [M]  /var/lib/dkms/alsa-firewire/3.11/build/sound/firewire/packets-buffer.o
  CC [M]  /var/lib/dkms/alsa-firewire/3.11/build/sound/firewire/fcp.o
  CC [M]  /var/lib/dkms/alsa-firewire/3.11/build/sound/firewire/cmp.o
  CC [M]  /var/lib/dkms/alsa-firewire/3.11/build/sound/firewire/amdtp.o
/var/lib/dkms/alsa-firewire/3.11/build/sound/firewire/amdtp.c: In function ‘amdtp_stream_pcm_abort’:
/var/lib/dkms/alsa-firewire/3.11/build/sound/firewire/amdtp.c:1067:3: error: implicit declaration of function ‘snd_pcm_stop_xrun’ [-Werror=implicit-function-declaration]
   snd_pcm_stop_xrun(pcm);
   ^
cc1: some warnings being treated as errors
scripts/Makefile.build:257: recipe for target '/var/lib/dkms/alsa-firewire/3.11/build/sound/firewire/amdtp.o' failed
make[3]: *** [/var/lib/dkms/alsa-firewire/3.11/build/sound/firewire/amdtp.o] Error 1
scripts/Makefile.build:402: recipe for target '/var/lib/dkms/alsa-firewire/3.11/build/sound/firewire' failed
make[2]: *** [/var/lib/dkms/alsa-firewire/3.11/build/sound/firewire] Error 2
Makefile:1381: recipe for target '_module_/var/lib/dkms/alsa-firewire/3.11/build' failed
make[1]: *** [_module_/var/lib/dkms/alsa-firewire/3.11/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/3.18.3-1-MANJARO/build'
Makefile:16: recipe for target 'all' failed
make: *** [all] Error 2

Cheers, Simon

takaswie, 2015-01-29:
Remove '#ifndef SNDRV_PCM_FORMAT_DSD_U32_LE' and corresponding '#endif' in backport.h, then your compiler will find the symbol 'snd_pcm_stop_xrun()':
https://github.com/takaswie/snd-firewire-improve/blob/master/backport.h

takaswie, 2015-01-29:
I can regenerate this issue in Ubuntu 15.04 userland.

# make -C /usr/src/linux-headers-3.18.0-11-generic/ M=`pwd`
make: Entering directory '/usr/src/linux-headers-3.18.0-11-generic'
  LD      /root/snd-firewire-improve/built-in.o
  LD      /root/snd-firewire-improve/sound/firewire/built-in.o
  CC [M]  /root/snd-firewire-improve/sound/firewire/lib.o
  CC [M]  /root/snd-firewire-improve/sound/firewire/iso-resources.o
  CC [M]  /root/snd-firewire-improve/sound/firewire/packets-buffer.o
  CC [M]  /root/snd-firewire-improve/sound/firewire/fcp.o
  CC [M]  /root/snd-firewire-improve/sound/firewire/cmp.o
  CC [M]  /root/snd-firewire-improve/sound/firewire/amdtp.o
/root/snd-firewire-improve/sound/firewire/amdtp.c: In function 'amdtp_stream_pcm_abort':
/root/snd-firewire-improve/sound/firewire/amdtp.c:1067:3: error: implicit declaration of function 'snd_pcm_stop_xrun' [-Werror=implicit-function-declaration]
   snd_pcm_stop_xrun(pcm);
   ^
cc1: some warnings being treated as errors
scripts/Makefile.build:257: recipe for target '/root/snd-firewire-improve/sound/firewire/amdtp.o' failed
make[2]: *** [/root/snd-firewire-improve/sound/firewire/amdtp.o] Error 1
scripts/Makefile.build:402: recipe for target '/root/snd-firewire-improve/sound/firewire' failed
make[1]: *** [/root/snd-firewire-improve/sound/firewire] Error 2
Makefile:1393: recipe for target '_module_/root/snd-firewire-improve' failed
make: *** [_module_/root/snd-firewire-improve] Error 2
make: Leaving directory '/usr/src/linux-headers-3.18.0-11-generic'

TODO: fix it.

takaswie, 2015-01-29:
Fix released.
https://github.com/takaswie/snd-firewire-improve/commit/c3f07d27ddf5f178054b896ab5c2961ecbe80211

quietpinguin, 2015-01-29:
Hi Takashi

Perfect - in every sense of the word!

- The modules built perfectly.
- The Focusrite Saffire PRO 24 is detected on-plug in.
- Using QjackCtl, jackd started and run correctly using alsa as the driver and PRO 24 as the interface.
- All the PRO 24 audio ins and outs could be routed to/from Ardour3.
- Ardour3 records unendlessly without a single XRUN.

This is truly a cool piece of programming...

Will there be a generic mixer GUI for dice based audio interfaces in alsa in the near future?

Just one question regarding the build process:
- Can / should I remove the symlink in the /usr/src/ folder created during build?

Thank you very much for your quick and excellent support, Simon

closed 2015-01-31 by takaswie