takaswie/linux-firewire-dkms #12: Firewire control transactions to ProjectMix I/O timing out.

wornwinter, 2015-01-31:
Hi,

Takashi, I remember talking to you about this before, but just wanted to highlight it again.

It seems that a firewire bus reset is necessary before sending commands to the ProjectMix in order for these commands to not keep timing out.

I realise that it may be worthwhile to include this in FFADO as opposed to your project here, but I thought for simplicity, including it in your driver may be a good decision.

Other than that, the driver is working great. It's allowed me to switch to Linux nearly full time by running a jack sink from pulseaudio. Good work! I really appreciate it.

Regards,
Darren

wornwinter, 2015-01-31:
I've created a pull request which solves this issue:

https://github.com/takaswie/snd-firewire-improve/pull/13

takaswie, 2015-01-31:
Hi Darren,

The prototype is:
int snd_bebob_maudio_special_discover(struct snd_bebob *bebob, bool is1814);

The value of boolean is used to switch Firewire1814 and ProjectMix I/O, therefore unrelated to generating bus-reset.

Actually, the snd-bebob executes one command before generating bus-reset.
 * FireWire bus driver
 * (the device is recognized.)
 * ->bebob_probe()
 *   ->snd_bebob_maudio_special_discover()
 *     ->avc_maudio_set_special_clk()
 *       ->fcp_avc_transaction() - command execution
 *   ->fw_schedule_bus_reset()
 * (bus-reset occurs)
 * ->bebob_update()
 *   ->snd_card_register()
 *   (ALSA character devices are added)

But this is not avoidable because of the way in which the FireWire bus driver handles drivers. I cannot program the code generating bus-reset before executing the command.

Regards

wornwinter, 2015-01-31:
Hi Takashi,

Stangely though, this change seems to not break the functionality of the soundcard, and it makes the ffado-mixer controls usable, whereas before they barely worked. I think we'll need to look more into this to find out exactly why.

Have you any ideas of why this fixes the issue?

Regards

takaswie, 2015-01-31:
> I think we'll need to look more into this to find out exactly why.

I think so, too. Sorry to trouble you...

Well, please clarify issues which you can see. The usability of ffado-mixer is just one? Are there other issues such as ALSA streaming works bad or you cannot hear sounds from your device? Or others?

Regards

wornwinter, 2015-01-31:
The streaming works great. Can't fault it.

The issue is that if I set is1814 to false, the ffado-mixer controls are unusable. It seems that control packets are not sending properly. If I set it to true, ffado-mixer works great.

Regards

takaswie, 2015-02-01:
> The issue is that if I set is1814 to false, the ffado-mixer controls are unusable.
> It seems that control packets are not sending properly. If I set it to true, ffado-mixer works great.

I require some logs from ffado-dbus-server. Please run 'ffado-dbus-server -v 8', then run ffado-mixer and change some control elements. Post the logs to alsa-devel or ffado-devel.

Regards

wornwinter, 2015-02-02:
The issue seems to have resolved itself. I'll post the logs if it starts to happen again. I think this should be considered closed for now.

closed 2015-02-02 by wornwinter