takaswie/linux-firewire-dkms #16: New digimagic (003amdtp) driver integration

zamaudio, 2015-03-21:
There was a bug in the implementation of digimagic, which has now been corrected in 003amdtp at http://github.com/x42/003amdtp .
Now we need to integrate it correctly into the digi00x driver. This may involve a slight change in how the digimagic state is stored and allocated.

zamaudio, 2015-03-24:
I tested commit 3eb7ec3440 with 003R+ and the digimagic is working on all channels. Thanks Takashi!

takaswie, 2015-03-27:
I think it better to report it to alsa-devel so that the other ALSA developers acknowledge it. The aim of our work is to merge this driver into upstream. To achieve this aim, we should have enough discussion on the place where the other ALSA developers work.

Regards

closed 2016-01-31 by takaswie