takaswie/linux-firewire-dkms #17: Discovered internal mixer controls - digi00x

zamaudio, 2015-03-24:
Hi Takashi,
I found the following controls for internal mixer matrix:

```
// Value 0 = off
// Value 1 = on
#define R003_MIX_CONTROL        0xffffe0000124ull

#define R003_MIX                0xffffe0000300ull
#define R003_MIX_ANALOG_1L      (0x00 | R003_MIX)
#define R003_MIX_ANALOG_1R      (0x04 | R003_MIX) 
#define R003_MIX_ANALOG_2L      (0x08 | R003_MIX)
#define R003_MIX_ANALOG_2R      (0x0c | R003_MIX)
#define R003_MIX_ANALOG_3L      (0x10 | R003_MIX)
#define R003_MIX_ANALOG_3R      (0x14 | R003_MIX)
#define R003_MIX_ANALOG_4L      (0x18 | R003_MIX)
#define R003_MIX_ANALOG_4R      (0x1c | R003_MIX)
#define R003_MIX_ANALOG_5L      (0x20 | R003_MIX)
#define R003_MIX_ANALOG_5R      (0x24 | R003_MIX)
#define R003_MIX_ANALOG_6L      (0x28 | R003_MIX)
#define R003_MIX_ANALOG_6R      (0x2c | R003_MIX)
#define R003_MIX_ANALOG_7L      (0x30 | R003_MIX)
#define R003_MIX_ANALOG_7R      (0x34 | R003_MIX)
#define R003_MIX_ANALOG_8L      (0x38 | R003_MIX)
#define R003_MIX_ANALOG_8R      (0x3c | R003_MIX)
#define R003_MIX_SPDIF_1L       (0x40 | R003_MIX)
#define R003_MIX_SPDIF_1R       (0x44 | R003_MIX)
#define R003_MIX_SPDIF_2L       (0x48 | R003_MIX)
#define R003_MIX_SPDIF_2R       (0x4c | R003_MIX)
#define R003_MIX_ADAT_1L        (0x50 | R003_MIX)
#define R003_MIX_ADAT_1R        (0x54 | R003_MIX)
#define R003_MIX_ADAT_2L        (0x58 | R003_MIX)
#define R003_MIX_ADAT_2R        (0x5c | R003_MIX)
#define R003_MIX_ADAT_3L        (0x60 | R003_MIX)
#define R003_MIX_ADAT_3R        (0x64 | R003_MIX)
#define R003_MIX_ADAT_4L        (0x68 | R003_MIX)
#define R003_MIX_ADAT_4R        (0x6c | R003_MIX)
#define R003_MIX_ADAT_5L        (0x70 | R003_MIX)
#define R003_MIX_ADAT_5R        (0x74 | R003_MIX)
#define R003_MIX_ADAT_6L        (0x78 | R003_MIX)
#define R003_MIX_ADAT_6R        (0x7c | R003_MIX)
#define R003_MIX_ADAT_7L        (0x80 | R003_MIX)
#define R003_MIX_ADAT_7R        (0x84 | R003_MIX)
#define R003_MIX_ADAT_8L        (0x88 | R003_MIX)
#define R003_MIX_ADAT_8R        (0x8c | R003_MIX)

#define R003_MIX_NONE           0x00000000
#define R003_MIX_1_TO_STEREO    0x18000000
#define R003_MIX_1_TO_1         0x20000000
```

takaswie, 2015-03-27:
Thanks. But I have never tested these functionality because no idea to implement it in usespace.
By the way, are there some registers to control gain of each analog input or output?

Regards

zamaudio, 2015-03-27:
No gain registers that I know of.  Besides there are knobs on the panel for that.

closed 2016-01-31 by takaswie