takaswie/linux-firewire-dkms #19: trying with audiofire2 on kernel 4.2 (mint)

ungentilgarcon, 2015-11-16:
hello, I it installed successfully, though I still have this
Failed to load module "module-alsa-card" (argument: "device_id="0" name="firewire-0x0014860fc44a8c55" card_name="alsa_card.firewire-0x0014860fc44a8c55" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=no card_properties="module-udev-detect.discovered=1""): initialization failed.

full dump here:
https://gist.github.com/ungentilgarcon/7fce5ee104c0cabe05af

If I modprobe I don't have a snd-fireworks, does that mean it didn't install itself?

I'm ready to do much to help make it work.

Audiofire firmware was downgraded to 4.8, oh and it works ok with jack.

regards

greg

ungentilgarcon, 2015-11-16:
here is the result of dkms installation,
https://gist.github.com/ungentilgarcon/1edde333e738024d9b62

ungentilgarcon, 2015-11-16:
here is the result of pactl
https://gist.github.com/ungentilgarcon/ffe54945fab09830bb71

no audiofire listed

takaswie, 2015-11-16:
Your AudioFire2 has 6 PCM channels for playback and 4 PCM channels for capture. PulseAudio 6.0 or former cannot handle the difference correctly. This bug is already fixed by this patch:
http://cgit.freedesktop.org/pulseaudio/pulseaudio/commit/?id=02dc6d84edde7af86e155b224b6aa98080ef2979

If you want to use AudioFire2 with PulseAudio, please load 'module-alsa-sink' or 'module-alsa-source' by your hand, like:
$ pactl load-module module-alsa-sink device=hw:0,4

closed 2016-01-31 by takaswie