takaswie/linux-firewire-dkms #20: Fireface 800 modprobe issue

brittyazel, 2017-06-30:
Hello,

My lab has a fireface 800 that we use for very timing sensitive audio output. We are using Ubuntu 17.04, Linux 4.10, and I downloaded your repository and followed the steps to install the modules via DKMS. The install is successfull, but when I execute "sudo modprobe snd-fireface" I get an error saying that it cannot find "amdtp_stream_pcm_ack" (I don't know the exact error, as I am not at that computer at the moment).

Any idea what I can do to get this error resolved so that I can use alsa for my fireface 800?

takaswie, 2017-06-30:
You might see below warnings when executing dkms command line.

> snd-fireface.ko:
> Running module version sanity check.
> Error! Module version 03F360D12AF05E4AD74EDD4 for snd-fireface.ko
> is not newer than what is already found in kernel 4.12.0-rc2-falcot (3421437241FDEA8742A9C82).
> You may override by specifying --force.

By following to the guidance, please add '--force' option to the command line.

brittyazel, 2017-06-30:
That isn't the problem. I can execute the dkms comand line fine, and I do see those errors, but the issue is when I actually try to load the snd-fireface module. It throws an unsatisfied reference to "amdtp_stream_pcm_ack" and says check my dmesg for details

takaswie, 2017-06-30:
You encounter symbol mismatch between snd-firewire-lib and your fireface. Please follow the above guidance.

brittyazel, 2017-06-30:
Yes, it is still not working. 

```
britt@DSS04624-PresentationPC:~/Downloads$ sudo dkms install alsa-firewire/4.4 --force

Creating symlink /var/lib/dkms/alsa-firewire/4.4/source ->
                 /usr/src/alsa-firewire-4.4

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=4.10.0-26-generic all....
cleaning build area...

DKMS: build completed.

snd-firewire-lib.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-26-generic/updates/dkms/

snd-isight.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-26-generic/updates/dkms/

snd-dice.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-26-generic/updates/dkms/

snd-fireworks.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-26-generic/updates/dkms/

snd-bebob.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-26-generic/updates/dkms/

snd-oxfw.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-26-generic/updates/dkms/

snd-firewire-digi00x.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-26-generic/updates/dkms/

snd-firewire-tascam.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-26-generic/updates/dkms/

snd-firewire-motu.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-26-generic/updates/dkms/

snd-fireface.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-26-generic/updates/dkms/

depmod...

DKMS: install completed.
```
```
britt@DSS04624-PresentationPC:~/Downloads$ sudo modprobe snd-fireface
modprobe: ERROR: could not insert 'snd_fireface': Unknown symbol in module, or unknown parameter (see dmesg)
```
`[18220.890017] snd_fireface: Unknown symbol amdtp_stream_pcm_ack (err 0)`

brittyazel, 2017-06-30:
nvm rebooting seemed to fix it

closed 2017-06-30 by brittyazel

ML-TakashiSAKAMOTO, 2017-06-30:
You might have missed  to unload relevant modules, before replacing modules by dkms.