takaswie/linux-firewire-dkms #25: snd-bebob : from kernel 4.13 to 5.3.19 and .20

jpargudo, 2019-10-30:
Hi there,

First, I'm really not sure this is the right place to write an Issue, but I didnt find better on the net, still waiting alsa-devel ML approval... :/

I upgraded my Ubuntu Disco (kernel 4.13) to Ubuntu Eoan (kernel 5.3.19) and I can report my SaffireLE / Firewire doesnt work anymore.

At startup it lights green ok, but no sound is playable, then the lights turn orange (like it is when it's not working), I hear the relay sound, then, the Saffire LE disapears from the sound menu.

I can see this in dmesg:

```
[   19.083583] snd-bebob fw1.0: Detect discontinuity of CIP: 10 50
[   19.746665] snd-bebob fw1.0: Detect discontinuity of CIP: A0 A8
[   19.846775] snd-bebob fw1.0: Detect discontinuity of CIP: 70 B0
[   20.335861] snd-bebob fw1.0: Detect discontinuity of CIP: 90 D0
[   20.856301] snd-bebob fw1.0: Detect discontinuity of CIP: 78 B8
[   21.312770] snd-bebob fw1.0: Detect discontinuity of CIP: 30 70
[   21.796716] snd-bebob fw1.0: Detect discontinuity of CIP: 00 40
[   22.347664] snd-bebob fw1.0: Detect discontinuity of CIP: 00 40
[   22.818247] snd-bebob fw1.0: Detect discontinuity of CIP: A8 E8
[   23.306079] snd-bebob fw1.0: Detect discontinuity of CIP: 98 D8
[   29.544652] snd-bebob fw1.0: Detect discontinuity of CIP: 10 50
[   30.272090] snd-bebob fw1.0: Detect discontinuity of CIP: A8 E8
[   30.776168] snd-bebob fw1.0: Detect discontinuity of CIP: 00 40
[   31.268123] snd-bebob fw1.0: Detect discontinuity of CIP: E0 20
[   31.766085] snd-bebob fw1.0: Detect discontinuity of CIP: E0 20
[   32.267903] snd-bebob fw1.0: Detect discontinuity of CIP: 88 C8
[   32.767989] snd-bebob fw1.0: Detect discontinuity of CIP: F8 38
[   33.263808] snd-bebob fw1.0: Detect discontinuity of CIP: 78 B8
[   33.757156] snd-bebob fw1.0: Detect discontinuity of CIP: 08 48
[   34.258860] snd-bebob fw1.0: Detect discontinuity of CIP: 80 C0
[   34.755571] snd-bebob fw1.0: Detect discontinuity of CIP: 40 80
[   35.261252] snd-bebob fw1.0: Detect discontinuity of CIP: 60 A0
[   35.753980] snd-bebob fw1.0: Detect discontinuity of CIP: 78 B8
[   36.253547] snd-bebob fw1.0: Detect discontinuity of CIP: C0 00
[   36.752500] snd-bebob fw1.0: Detect discontinuity of CIP: 08 48
[   37.255203] snd-bebob fw1.0: Detect discontinuity of CIP: B0 F0
[  277.760550] snd-bebob fw1.0: Detect discontinuity of CIP: 40 80
[  278.488972] snd-bebob fw1.0: Detect discontinuity of CIP: C0 00
[  278.986290] snd-bebob fw1.0: Detect discontinuity of CIP: E0 20
[  279.485998] snd-bebob fw1.0: Detect discontinuity of CIP: 18 58
[  279.983838] snd-bebob fw1.0: Detect discontinuity of CIP: D0 10
[  280.486776] snd-bebob fw1.0: Detect discontinuity of CIP: C0 00
[  280.979487] snd-bebob fw1.0: Detect discontinuity of CIP: 30 70
[  281.480952] snd-bebob fw1.0: Detect discontinuity of CIP: 78 B8
[  281.975413] snd-bebob fw1.0: Detect discontinuity of CIP: 38 78
[  282.475487] snd-bebob fw1.0: Detect discontinuity of CIP: 60 A0
[  282.979446] snd-bebob fw1.0: Detect discontinuity of CIP: 88 C8
[  283.472403] snd-bebob fw1.0: Detect discontinuity of CIP: 88 C8
[  283.968484] snd-bebob fw1.0: Detect discontinuity of CIP: 68 A8
[  284.471585] snd-bebob fw1.0: Detect discontinuity of CIP: 30 70
[  284.562886] usb 3-4.1: reset high-speed USB device number 5 using xhci_hcd
[  284.965508] snd-bebob fw1.0: Detect discontinuity of CIP: D0 10
[  285.469348] snd-bebob fw1.0: Detect discontinuity of CIP: 68 A8
[  285.965174] snd-bebob fw1.0: Detect discontinuity of CIP: 20 60
[  285.981618] firewire_core 0000:05:01.0: phy config: new root=ffc1, gap_count=5
[  290.103982] firewire_core 0000:05:01.0: phy config: new root=ffc1, gap_count=5
```

I tried update the kernel with eoan-proposed I know run 5.3.20 and I have the same problem (the trace above is made with 5.3.20.

So please, first: tell me where to report this properly to the proper team... ?

And second, If you have any tools or tips to report better, with a trace on or something?

Thanks a lot !!!!!

takaswie, 2019-10-30:
Hi,

Thanks for your report. I did reply in the list:
https://mailman.alsa-project.org/pipermail/alsa-devel/2019-October/157741.html

jpargudo, 2019-10-30:
Hi !

Thanks a lot! 

Let's close this issue here then?

takaswie, 2019-11-26:
Hi @jpargudo ,

I'd like you to test the latest driver by backporting it into your system, and report your result.
https://mailman.alsa-project.org/pipermail/alsa-devel/2019-November/158294.html

I maintain this repository for this purpose ;)

Thanks

muziker, 2020-03-05:
Hi there, i'd like to confirm that once i 
1. in qjackctl, select alsa as the driver
2. Select hw:saffireLE as the interface
3. using 5.3.0-18-lowlatency on ubuntu studio

I get:
`
kernel: snd-bebob fw1.0: Detect discontinuity of CIP: 10 E8`
`...`
`kernel: Call Trace`
`...`
`kernel: Code: Bad RIP value
`
I can use
1. in qjackctl, select firewire
But there are frequent dropouts

muziker, 2020-03-05:
I've used the current master HEAD, and it seems to be working

muziker, 2020-03-05:
Tried other settings frames/period + periods/buffer and it seems to work ok. The performance isn't particularly stellar, and cannot reach <10msec. There's still some xruns, once every 10 - 20 minutes.

muziker, 2020-03-06:
Might not be related, but everytime qjackctl freezes, i'll have to delete the /dev/shm of jack shared memory files

muziker, 2020-03-06:
From reading all the issues, it really does seem like the discontinuity in packet streaming eventually leads to a large enough discrepancy that jack using alsa firewire stops working. 

https://lore.kernel.org/alsa-devel/ad6f8c036538aa755017efe976ac223bb7c90be3.camel@argudo.org/t/

I'm not sure how he does it, but it's definitely not 10msec latency that works

jpargudo, 2020-03-06:
Hi there,

Le jeudi 05 mars 2020 à 18:51 -0800, muziker a écrit :
> From reading all the issues, it really does seem like the
> discontinuity in packet streaming eventually leads to a large enough
> discrepancy that jack using alsa firewire stops working. [...]

FYI, since that problem I got rid of firewire and the Saffire LE,
waiting for better days :-( ...

I can put it back in the box and continue debugging for you if you
want. Just tell me how I could help there :/

I think also of putting this in a comparable box I have here, and give
you ssh access if you want?...

Just tell me !

Cheers,

-- 
Jean-Paul

muziker, 2020-03-06:
Well, i was hoping you'd have the settings that would solve all the issues. After a lengthy trial by error, i've managed to come to this setting, that so far seems to work, though it still gives a xrun every few minutes or so, they are usually inaudible. Be advised that my machine is one of those macbooks with all the connectors on one side, which means that they might be somehow sharing either a powerline, or some bus controller, which can affect performance. 

At first i thought it was a problem with the irqs, but i found using this repo, with alsa, 48k, frames/period 256, periods/buffer 6, non realtime, no verbose messages, no memory lock , and no midi driver, to give a latency of 32 msec. The general issue is probably the cip discontinuity and timing information. I also found some annoying interactions between jackd and the qjackctl settings, in that, to fully ensure the correct values are used, all jack processes must be killed. Also i've disabled pulseaudio permanently on boot, but will have it running later. Though to be fully thorough, the machine needs to be rebooted every change in setting, just to be certain. I didn't do that all the time, relying on pkill to get rid of jack, and ffado-test BusReset to reset the controller. 

I really don't know how windows and mac gets this interface running at low latencies.

takaswie, 2021-05-24:
I attempt to implement media clock recovery. If still interested, please test current HEAD of `topic/media-clock-recovery` remote branch (3a2c5fd52657).

takaswie, 2021-06-01:
I note that patchset is posted to upstream:

* https://lore.kernel.org/alsa-devel/20210601081753.9191-1-o-takashi@sakamocchi.jp/

jpargudo, 2021-06-01:
Thank you @takaswie 

I'll try asap.. And give feedback.

takaswie, 2021-06-01:
@jpargudo 

> I'll try asap.. And give feedback.

If you try, it's convenient to use`topic/media-clock-recovery` in this repository.

Thanks

takaswie, 2021-06-01:
The patchset is merged to upstream. The patches are also available in master branch of the repository.

takaswie, 2021-06-11:
Let me close the issue since no reactions during a recent 10 days.

closed 2021-06-11 by takaswie