takaswie/linux-firewire-dkms #28: Can't compile module with latest fedora kernel

danielmellado, 2020-06-19:
Hi, I'm trying to compile the latest module with fedora32 and dkms fails

`$ sudo dkms install alsa-firewire/4.17

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.6.18-300.fc32.x86_64 all...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.6.18-300.fc32.x86_64 (x86_64)
Consult /var/lib/dkms/alsa-firewire/4.17/build/make.log for more information.
`

Checking the file shows as follows:
`DKMS make.log for alsa-firewire-4.17 for kernel 5.6.18-300.fc32.x86_64 (x86_64)
vie 19 jun 2020 10:09:18 CEST
make -C /lib/modules/5.6.18-300.fc32.x86_64/build M=/var/lib/dkms/alsa-firewire/4.17/build modules
make[1]: Entering directory '/usr/src/kernels/5.6.18-300.fc32.x86_64'
  CC [M]  /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/dice/dice-transaction.o
  CC [M]  /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/bebob/bebob_command.o
  CC [M]  /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/fireface/ff.o
  CC [M]  /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/digi00x/amdtp-dot.o
  CC [M]  /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/tascam/tascam-proc.o
  CC [M]  /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/oxfw-command.o
  CC [M]  /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/fireworks/fireworks_transaction.o
  CC [M]  /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/motu/motu.o
In file included from /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../../backport.h:2,
                 from /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/oxfw.h:19,
                 from /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/oxfw-command.c:8:
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:460:18: error: field ‘trigger_tstamp’ has incomplete type
  460 |  struct timespec trigger_tstamp; /* time when stream was started/stopped/paused */
      |                  ^~~~~~~~~~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:461:18: error: field ‘tstamp’ has incomplete type
  461 |  struct timespec tstamp;  /* reference timestamp */
      |                  ^~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:470:18: error: field ‘audio_tstamp’ has incomplete type
  470 |  struct timespec audio_tstamp; /* sample counter, wall clock, PHC or on-demand sync'ed */
      |                  ^~~~~~~~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:471:18: error: field ‘driver_tstamp’ has incomplete type
  471 |  struct timespec driver_tstamp; /* useful in case reference system tstamp is reported with delay */
      |                  ^~~~~~~~~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:473:37: error: invalid application of ‘sizeof’ to incomplete type ‘struct timespec’
  473 |  unsigned char reserved[52-2*sizeof(struct timespec)]; /* must be filled with zero */
      |                                     ^~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:480:18: error: field ‘tstamp’ has incomplete type
  480 |  struct timespec tstamp;  /* Timestamp */
      |                  ^~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:482:18: error: field ‘audio_tstamp’ has incomplete type
  482 |  struct timespec audio_tstamp; /* from sample counter or wall clock */
      |                  ^~~~~~~~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:652:18: error: field ‘tstamp’ has incomplete type
  652 |  struct timespec tstamp;  /* Timestamp */
      |                  ^~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:764:18: error: field ‘tstamp’ has incomplete type
  764 |  struct timespec tstamp;  /* Timestamp - last update */
      |                  ^~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:814:18: error: field ‘tstamp’ has incomplete type
  814 |  struct timespec tstamp;
      |                  ^~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:957:18: error: field ‘tstamp’ has incomplete type
  957 |  struct timespec tstamp;
      |                  ^~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/oxfw/../../.././include/uapi/sound/asound.h:958:36: error: invalid application of ‘sizeof’ to incomplete type ‘struct timespec’
  958 |  unsigned char reserved[128-sizeof(struct timespec)];
      |                                    ^~~~~~
In file included from /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/tascam/./../../../backport.h:2,
                 from /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/tascam/./tascam.h:22,
                 from /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/tascam/tascam-proc.c:8:
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/tascam/./../../.././include/uapi/sound/asound.h:460:18: error: field ‘trigger_tstamp’ has incomplete type
  460 |  struct timespec trigger_tstamp; /* time when stream was started/stopped/paused */
[... 560 lines cut ...]
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/digi00x/../../.././include/uapi/sound/asound.h:652:18: error: field ‘tstamp’ has incomplete type
  652 |  struct timespec tstamp;  /* Timestamp */
      |                  ^~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/digi00x/../../.././include/uapi/sound/asound.h:764:18: error: field ‘tstamp’ has incomplete type
  764 |  struct timespec tstamp;  /* Timestamp - last update */
      |                  ^~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/digi00x/../../.././include/uapi/sound/asound.h:814:18: error: field ‘tstamp’ has incomplete type
  814 |  struct timespec tstamp;
      |                  ^~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/digi00x/../../.././include/uapi/sound/asound.h:957:18: error: field ‘tstamp’ has incomplete type
  957 |  struct timespec tstamp;
      |                  ^~~~~~
/var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/digi00x/../../.././include/uapi/sound/asound.h:958:36: error: invalid application of ‘sizeof’ to incomplete type ‘struct timespec’
  958 |  unsigned char reserved[128-sizeof(struct timespec)];
      |                                    ^~~~~~
make[4]: *** [scripts/Makefile.build:267: /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/digi00x/digi00x-midi.o] Error 1
make[3]: *** [scripts/Makefile.build:505: /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/digi00x] Error 2
make[2]: *** [scripts/Makefile.build:505: /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire] Error 2
make[1]: *** [Makefile:1695: /var/lib/dkms/alsa-firewire/4.17/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.6.18-300.fc32.x86_64'
make: *** [Makefile:23: all] Error 2
`
As I'm not familiar with this process, I'm not sure if I might be missing any required library, mind having a look at this?

Thanks!

takaswie, 2020-06-19:
Hi,

Thanks for your report and I'm sorry for the inconvenience.

Recently Linux kernel got changes for year 2038 issue and UAPI header of Linux sound subsystem was changed. On the other hand, in my repository, old headers are included. This is the cause of issue.

takaswie, 2020-06-19:
I'll try to backport patches including the changes.

takaswie, 2020-06-19:
```
$ git log --oneline v5.7 include/uapi/sound/asound.h include/uapi/sound/asequencer.h
46b770f720bd ALSA: uapi: Fix sparse warning
a103a3989993 ALSA: control: Fix incompatible protocol error
bd3eb4e87eb3 ALSA: ctl: bump protocol version up to v2.1.0
ff16351e3f30 ALSA: ctl: remove dimen member from elem_info structure
542283566679 ALSA: ctl: remove unused macro for timestamping of elem_value
7fd7d6c50451 ALSA: uapi: Fix typos and header inclusion in asound.h
1cfaef961703 ALSA: bump uapi version numbers
80fe7430c708 ALSA: add new 32-bit layout for snd_pcm_mmap_status/control
07094ae6f952 ALSA: Avoid using timespec for struct snd_timer_tread
d9e5582c4bb2 ALSA: Avoid using timespec for struct snd_rawmidi_status
3ddee7f88aaf ALSA: Avoid using timespec for struct snd_pcm_status
a4e7dd35b9da ALSA: Avoid using timespec for struct snd_ctl_elem_value
a07804cc7472 ALSA: Avoid using timespec for struct snd_timer_status
b5bdbb6ccd11 ALSA: uapi: #include <time.h> in asound.h
a98401518def ALSA: timer: fix wrong comment to refer to 'SNDRV_TIMER_PSFLG_*'
...
```

As long as I investigated, the above patches should be backported.

takaswie, 2020-06-19:
@danielmellado Would you please try HEAD of `topic/backport-y2038-changes` branch and report the result?

danielmellado, 2020-06-19:
Now it did work and compile just fine, thanks for your quick response!

danielmellado, 2020-06-19:
In any case, I've another doubt, which is outside of the scope of this bug, so will probably just cc alsa-devel, but after modprobing `snd-firewire-lib` and `snd-firewire-motu` and doing `alsactl restore` I would've expected to see my MOTU 896HD at /proc/asound/cards. Am I missing some usage of the modules? Thanks in advance!

takaswie, 2020-06-19:
> I would've expected to see my MOTU 896HD at `/proc/asound/cards`.

MOTU 896HD is supported neither in this repository[1] nor upstream[2], since I've had no chance to access to the device (it's still expensive even if enough later being discontinued).

If I expect your cooperation, I will work to support it. However, current ALSA firewire-motu driver still has some issues about media clock recovery and it's likely that I cannot meet your request fully.

[1] https://github.com/takaswie/snd-firewire-improve/blob/master/sound/firewire/motu/motu.c#L186
[2] https://git.kernel.org/pub/scm/linux/kernel/git/tiwai/sound.git/tree/sound/firewire/motu/motu.c?h=for-next#n186

Regards

danielmellado, 2020-06-19:
Hi @takaswie, yeah, that'd be awesome! Let me know how I can help you achieve that! For now I've got the device working with libffado + jack / pulseaudio bridge but it'd be really cool to have it under alsa. Regarding the device, feel free to reach me out in order to get any logs/info/tests you'd like me to run over it. Thanks a lot!

takaswie, 2020-06-19:
> Hi @takaswie, yeah, that'd be awesome! Let me know how I can help you achieve that! For now
> I've got the device working with libffado + jack / pulseaudio bridge but it'd be really cool to have it [...]

Hm.

If you're satisfied to FFADO stack, I think it better just to use it, because it takes you a batch of free time and efforts to assist my work. In a point of life, time is finite and expensive. It's better to use the time for more important thing to you.

Novertheless you are going to assist my work, I'm glad to hear it and make some instructions to gather device data.

Thanks

danielmellado, 2020-06-26:
Hi @takaswie, I'd love seeing this working under alsa and I'm pretty sure that I can save some time to help you with that, so let me know on what should we start with ;)

takaswie, 2020-07-08:
Hi @danielmellado,

I'm sorry to be late for reply but I was pretty busy for my work of ALSA control services in userspace[1].

In a couple of days, I'm prepare python 3 script to retrieve data for your model. It's libhinawa[2] application and I'd like you to install the library with gobject-introspection[3] support.

[1] https://mailman.alsa-project.org/pipermail/alsa-devel/2020-July/170331.html
[2] https://github.com/alsa-project/libhinawa
[3] https://gi.readthedocs.io/en/latest/

Thanks for your patience.

takaswie, 2020-07-12:
Hi @danielmellado ,

> In a couple of days, I'm prepare python 3 script to retrieve data for your model. It's libhinawa[2] application and I'd like you to install the library with gobject-introspection[3] support in advance.

I forgot to note that [PyGObject](https://pygobject.readthedocs.io/en/latest/) is required to load libhinawa in Python 3 runtime via gobject introspection.

Well, I prepare for below Python 3 script to detect information required for packet streaming in 896HD case.

```
#!/usr/bin/env python3

import gi
gi.require_version('GLib', '2.0')
gi.require_version('Hinawa', '3.0')
from gi.repository import GLib, Hinawa

from struct import unpack
from threading import Thread

ADDR_BASE = 0xfffff0000000

RATES = (
    44100,
    48000,
    88200,
    96000,
    176400,
    192000,
)

def detect_clock_status(req, node):
    V2_CLOCK_STATUS_OFFSET = 0x0b14

    V2_CLOCK_RATE_MASK = 0x00000038
    V2_CLOCK_RATE_SHIFT = 3
    V2_CLOCK_SRC_MASK = 0x00000007
    V2_CLOCK_SRC_SHIFT = 0
    V2_CLOCK_FETCH_ENABLE = 0x02000000
    V2_CLOCK_MODEL_SPECIFIC = 0x04000000

    # Parse for rate of sampling clock.
    req = Hinawa.FwReq.new()
    frames = req.transaction(node, Hinawa.FwTcode.READ_QUADLET_REQUEST,
                             ADDR_BASE + V2_CLOCK_STATUS_OFFSET, 4, [0] * 4)
    val = unpack('>I', frames)[0]
    print('  0x0b14: {:08x}'.format(val))

    rate = (val & V2_CLOCK_RATE_MASK) >> V2_CLOCK_RATE_SHIFT
    if rate < len(RATES):
        print('    rate configuration: {}'.format(RATES[rate]))

    # Parse for source of sampling clock.
    src = (val & V2_CLOCK_SRC_MASK) >> V2_CLOCK_SRC_SHIFT;
    if src == 0:
        label = 'internal'
    elif src == 2:
        label = 'S/PDIF on coax'
    elif src == 4:
        label = 'Word on BNC'
    else:
        label = 'Unknown({})'.format(src)
    print('    src configuration: {}'.format(label))

def detect_io_conf(req, node):
    V2_IN_OUT_CONF_OFFSET = 0x0c04

    V2_OPT_OUT_IFACE_MASK = 0x00000c00
    V2_OPT_OUT_IFACE_SHIFT = 10
    V2_OPT_IN_IFACE_MASK = 0x00000300
    V2_OPT_IN_IFACE_SHIFT = 8

    V2_OPT_IFACE_MODES = (
        'None',
        'ADAT',
        'S/PDIF',
    )

    req = Hinawa.FwReq.new()
    frames = req.transaction(node, Hinawa.FwTcode.READ_QUADLET_REQUEST,
                             ADDR_BASE + V2_IN_OUT_CONF_OFFSET, 4, [0] * 4)
    val = unpack('>I', frames)[0]
    print('  0x0b14: {:08x}'.format(val))

    # Parse for optical output interface.
    opt_out_iface = (val & V2_OPT_OUT_IFACE_MASK) >> V2_OPT_OUT_IFACE_SHIFT
    if opt_out_iface < len(V2_OPT_IFACE_MODES):
        label = V2_OPT_IFACE_MODES[opt_out_iface]
    else:
        label = 'Unkown({})'.format(opt_out_iface)
    print('    opt out iface configuration: {}'.format(label))

    # Parse for optical input interface.
    opt_in_iface = (val & V2_OPT_IN_IFACE_MASK) >> V2_OPT_IN_IFACE_SHIFT
    if opt_in_iface < len(V2_OPT_IFACE_MODES):
        label = V2_OPT_IFACE_MODES[opt_in_iface]
    else:
        label = 'Unkown({})'.format(opt_in_iface)
    print('    opt in iface configuration: {}'.format(label))

unit = Hinawa.SndMotu.new()
unit.open('/dev/snd/hwC2D0')

ctx = GLib.MainContext.new()

src = unit.create_source()
src.attach(ctx)
src = unit.get_node().create_source()
src.attach(ctx)

dispatcher = GLib.MainLoop.new(ctx, False)
th = Thread(target=lambda d: d.run(), args=(dispatcher, ))
th.start()

req = Hinawa.FwReq.new()

print('Detect clock status:')
detect_clock_status(req, unit.get_node())

print('Detect I/O configuration:')
detect_io_conf(req, unit.get_node())

dispatcher.quit()
th.join()
```

When you have enough permissions for ALSA HwDep character device (`/dev/snd/hwC%uD%u`) and Linux FireWire character device (`/dev/fw%u`), executing the above Python 3 script brings below output:

```
$ python3 896hd-detect
Detect clock status:
  0x0b14: 04000000
    rate configuration: 44100
    src configuration: internal
Detect I/O configuration:
  0x0b14: 03000001
    opt out iface configuration: None
    opt in iface configuration: None
```

What I'd like you to check is the information about I/O configuration. Motu 896HD has some knobs to control internal configuration. By the knobs, you can change optical input/output configuration between `None` or `ADAT`. You can check whether the above script can detect the configuration correctly or not when changing the configuration by hand.

Cheers.

danielmellado, 2020-08-06:
Hi @takaswie, sorry for the delay as well! I ran into some issues when installing libhinawa. Although I got all the dependencies, I guess something is wrong there, after I do ninja install:

```
$ meson -Dgtk_doc=true . build
The Meson build system
Version: 0.55.0
Source dir: /home/dmellado/Devel/daniel/alsa/libhinawa
Build dir: /home/dmellado/Devel/daniel/alsa/libhinawa/build
Build type: native build
Using 'PKG_CONFIG_PATH' from environment with value: '/usr/lib64/pkgconfig:/usr/share/pkgconfig:/usr/local/lib64/pkgconfig'
Using 'PKG_CONFIG_PATH' from environment with value: '/usr/lib64/pkgconfig:/usr/share/pkgconfig:/usr/local/lib64/pkgconfig'
Project name: libhinawa
Project version: 2.0.0
C compiler for the host machine: cc (gcc 10.2.1 "cc (GCC) 10.2.1 20200723 (Red Hat 10.2.1-1)")
C linker for the host machine: cc ld.bfd 2.34-4
Host machine cpu family: x86_64
Host machine cpu: x86_64
Has header "backport.h" : YES 
Configuring config.h using configuration
Found pkg-config: /usr/bin/pkg-config (1.6.3)
Using 'PKG_CONFIG_PATH' from environment with value: '/usr/lib64/pkgconfig:/usr/share/pkgconfig:/usr/local/lib64/pkgconfig'
Run-time dependency gobject-2.0 found: YES 2.64.4
Found pkg-config: /usr/bin/pkg-config (1.6.3)
Using 'PKG_CONFIG_PATH' from environment with value: '/usr/lib64/pkgconfig:/usr/share/pkgconfig:/usr/local/lib64/pkgconfig'
Program glib-genmarshal found: YES
Program glib-mkenums found: YES
Program glib-mkenums found: YES
Using 'PKG_CONFIG_PATH' from environment with value: '/usr/lib64/pkgconfig:/usr/share/pkgconfig:/usr/local/lib64/pkgconfig'
Build-time dependency gobject-introspection-1.0 found: YES 1.64.1
Program g_ir_scanner found: YES (/usr/bin/g-ir-scanner)
Program g_ir_compiler found: YES (/usr/bin/g-ir-compiler)
Configuring version.xml using configuration
Program gtkdoc-scan found: YES
Program gtkdoc-scangobj found: YES
Program gtkdoc-mkdb found: YES
Program gtkdoc-mkhtml found: YES
Program gtkdoc-fixxref found: YES
Program fw-node found: YES
Program fw-req found: YES
Program fw-resp found: YES
Program fw-fcp found: YES
Program snd-unit found: YES
Program hinawa-enum found: YES
Program snd-dice found: YES
Program snd-efw found: YES
Program snd-dg00x found: YES
Program snd-motu found: YES
Program snd-tscm found: YES
Build targets in project: 9

Found ninja-1.10.0 at /usr/bin/ninja
```

```
$ ninja
[1/19] Generating hinawa_sigs_marshal_h with a custom command
INFO: Reading ../src/hinawa_sigs_marshal.list...
[2/19] Generating hinawa_sigs_marshal_c with a custom command
INFO: Reading ../src/hinawa_sigs_marshal.list...
[18/19] Generating Hinawa-3.0.gir with a custom command
g-ir-scanner: link: gcc -pthread -o /home/dmellado/Devel/daniel/alsa/libhinawa/build/tmp-introspect1h9kbibh/Hinawa-3.0 /home/dmellado/Devel/daniel/alsa/libhinawa/build/tmp-introspect1h9kbibh/Hinawa-3.0.o -L. -Wl,-rpath,. -Wl,--no-as-needed -L/home/dmellado/Devel/daniel/alsa/libhinawa/build/src -Wl,-rpath,/home/dmellado/Devel/daniel/alsa/libhinawa/build/src -lhinawa -lgobject-2.0 -lglib-2.0 -lgirepository-1.0 -lgio-2.0 -lgobject-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lglib-2.0 -lglib-2.0
[19/19] Generating Hinawa-3.0.typelib with a custom command
```
```
$sudo ninja install
[0/1] Installing files.
Installing src/hinawa_enums.h to /usr/local/include/libhinawa
Installing src/libhinawa.so.2.0.0 to /usr/local/lib64
Installing src/Hinawa-3.0.gir to /usr/local/share/gir-1.0
Installing src/Hinawa-3.0.typelib to /usr/local/lib64/girepository-1.0
[... 35 lines cut ...]
html/index-all.html:315: warning: no link for: "WITH-BACKPORT:CAPS" -> (WITH_BACKPORT).```
```

But when I run your script I get the following:

```
$ ./motu896hd.py 
Traceback (most recent call last):
  File "./motu896hd.py", line 5, in <module>
    gi.require_version('Hinawa', '3.0')
  File "/usr/lib64/python3.8/site-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Hinawa not available
```

I thought this could be related to some path and so so I even generated an rpm based on your instructions (just changing tar.gz to tar.xz, as it's the new format used now). If I go and install the rpm `libhinawa-2.0.0-1.fc32.x86_64.rpm` and `libhinawa-devel-2.0.0-1.fc32.x86_64.rpm` I got the following running the script.

```
$ ./motu896hd.py 
Traceback (most recent call last):
  File "./motu896hd.py", line 92, in <module>
    unit.open('/dev/snd/hwC2D0')
gi.repository.GLib.Error: "HinawaSndUnit": 220: No such file or directory (2)
```
That leads me to think that there's some issue on the path or something, but if you could lend me a hand on this it would be extremely helpful. Thanks!

danielmellado, 2020-08-06:
btw, wouldn't you be on freenode? maybe that'd make things easier ;)

danielmellado, 2020-08-06:
also, on `/dev/snd` I don't get to see any `hwC2D0' device, but 

```
[$ ls /dev/snd/
by-id  by-path  controlC0  controlC1  controlC2  hwC0D0  hwC0D2  pcmC0D0c  pcmC0D0p  pcmC0D10p  pcmC0D3p  pcmC0D7p  pcmC0D8p  pcmC0D9p  pcmC1D0c  pcmC1D0p  pcmC2D0c  seq  timer
```

danielmellado, 2020-08-06:
changing that to `hwC0D2` just in case, and it doesn't seem to be the proper one:

```
./motu896hd.py 
Traceback (most recent call last):
  File "./motu896hd.py", line 93, in <module>
    unit.open('/dev/snd/hwC0D2')
gi.repository.GLib.Error: "HinawaSndUnit": 226: Inappropriate ioctl for device (25)
```

takaswie, 2020-08-06:
Oops. How stupid I am...

```
$ ./motu896hd.py 
Traceback (most recent call last):
  File "./motu896hd.py", line 92, in <module>
    unit.open('/dev/snd/hwC2D0')
gi.repository.GLib.Error: "HinawaSndUnit": 220: No such file or directory (2)
```

ALSA firewire-motu driver doesn't handle your unit yet. So the instance should be from `HinawaFwNode` instead of `HinawaSndUnit`...

> btw, wouldn't you be on freenode? maybe that'd make things easier ;)

Would you please decide which channel in which we are going to log?

Thanks

takaswie, 2020-08-06:
This rewritten code should work for your HD896:

```
#!/usr/bin/env python3

import gi
gi.require_version('GLib', '2.0')
gi.require_version('Hinawa', '3.0')
from gi.repository import GLib, Hinawa

from struct import unpack
from threading import Thread

ADDR_BASE = 0xfffff0000000

RATES = (
    44100,
    48000,
    88200,
    96000,
    176400,
    192000,
)

def detect_clock_status(req, node):
    V2_CLOCK_STATUS_OFFSET = 0x0b14

    V2_CLOCK_RATE_MASK = 0x00000038
    V2_CLOCK_RATE_SHIFT = 3
    V2_CLOCK_SRC_MASK = 0x00000007
    V2_CLOCK_SRC_SHIFT = 0
    V2_CLOCK_FETCH_ENABLE = 0x02000000
    V2_CLOCK_MODEL_SPECIFIC = 0x04000000

    # Parse for rate of sampling clock.
    req = Hinawa.FwReq.new()
    frames = req.transaction(node, Hinawa.FwTcode.READ_QUADLET_REQUEST,
                             ADDR_BASE + V2_CLOCK_STATUS_OFFSET, 4, [0] * 4)
    val = unpack('>I', frames)[0]
    print('  0x0b14: {:08x}'.format(val))

    rate = (val & V2_CLOCK_RATE_MASK) >> V2_CLOCK_RATE_SHIFT
    if rate < len(RATES):
        print('    rate configuration: {}'.format(RATES[rate]))

    # Parse for source of sampling clock.
    src = (val & V2_CLOCK_SRC_MASK) >> V2_CLOCK_SRC_SHIFT;
    if src == 0:
        label = 'internal'
    elif src == 2:
        label = 'S/PDIF on coax'
    elif src == 4:
        label = 'Word on BNC'
    else:
        label = 'Unknown({})'.format(src)
    print('    src configuration: {}'.format(label))

def detect_io_conf(req, node):
    V2_IN_OUT_CONF_OFFSET = 0x0c04

    V2_OPT_OUT_IFACE_MASK = 0x00000c00
    V2_OPT_OUT_IFACE_SHIFT = 10
    V2_OPT_IN_IFACE_MASK = 0x00000300
    V2_OPT_IN_IFACE_SHIFT = 8

    V2_OPT_IFACE_MODES = (
        'None',
        'ADAT',
        'S/PDIF',
    )

    req = Hinawa.FwReq.new()
    frames = req.transaction(node, Hinawa.FwTcode.READ_QUADLET_REQUEST,
                             ADDR_BASE + V2_IN_OUT_CONF_OFFSET, 4, [0] * 4)
    val = unpack('>I', frames)[0]
    print('  0x0b14: {:08x}'.format(val))

    # Parse for optical output interface.
    opt_out_iface = (val & V2_OPT_OUT_IFACE_MASK) >> V2_OPT_OUT_IFACE_SHIFT
    if opt_out_iface < len(V2_OPT_IFACE_MODES):
        label = V2_OPT_IFACE_MODES[opt_out_iface]
    else:
        label = 'Unkown({})'.format(opt_out_iface)
    print('    opt out iface configuration: {}'.format(label))

    # Parse for optical input interface.
    opt_in_iface = (val & V2_OPT_IN_IFACE_MASK) >> V2_OPT_IN_IFACE_SHIFT
    if opt_in_iface < len(V2_OPT_IFACE_MODES):
        label = V2_OPT_IFACE_MODES[opt_in_iface]
    else:
        label = 'Unkown({})'.format(opt_in_iface)
    print('    opt in iface configuration: {}'.format(label))

node = Hinawa.FwNode.new()
node.open('/dev/fw1')

ctx = GLib.MainContext.new()

src = node.create_source()
src.attach(ctx)

dispatcher = GLib.MainLoop.new(ctx, False)
th = Thread(target=lambda d: d.run(), args=(dispatcher, ))
th.start()

req = Hinawa.FwReq.new()

print('Detect clock status:')
detect_clock_status(req, node)

print('Detect I/O configuration:')
detect_io_conf(req, node)

dispatcher.quit()
th.join()
```

The diff from the previous version:

```
$ diff -u prev motu896hd.py 
--- prev	2020-08-06 18:42:46.426353678 +0900
+++ motu896hd.py	2020-08-06 18:41:28.852895533 +0900
@@ -88,14 +88,12 @@
         label = 'Unkown({})'.format(opt_in_iface)
     print('    opt in iface configuration: {}'.format(label))
 
-unit = Hinawa.SndMotu.new()
-unit.open('/dev/snd/hwC2D0')
+node = Hinawa.FwNode.new()
+node.open('/dev/fw1')
 
 ctx = GLib.MainContext.new()
 
-src = unit.create_source()
-src.attach(ctx)
-src = unit.get_node().create_source()
+src = node.create_source()
 src.attach(ctx)
 
 dispatcher = GLib.MainLoop.new(ctx, False)
@@ -105,10 +103,10 @@
 req = Hinawa.FwReq.new()
 
 print('Detect clock status:')
-detect_clock_status(req, unit.get_node())
+detect_clock_status(req, node)
 
 print('Detect I/O configuration:')
-detect_io_conf(req, unit.get_node())
+detect_io_conf(req, node)
 
 dispatcher.quit()
 th.join()
```

Thanks

takaswie, 2020-08-06:
And let us move to another issue in which we discuss about MOTU 896HD support ;)

closed 2020-08-06 by takaswie