takaswie/linux-firewire-dkms #29: Add support for MOTU 896 HD

danielmellado, 2020-08-06:
Coming from https://github.com/takaswie/snd-firewire-improve/issues/28

danielmellado, 2020-08-06:
@takaswie this is the output from the script how ;)

```
$ ./motu896hd.py 
Detect clock status:
  0x0b14: 00000000
    rate configuration: 44100
    src configuration: internal
Detect I/O configuration:
  0x0b14: 03000006
    opt out iface configuration: None
    opt in iface configuration: None
```

In terms of the physical knobs, the do control monitor level/volume + mix 1/4 and meters, if pressed.

takaswie, 2020-08-06:
@danielmellado Great ;)

> In terms of the physical knobs, the do control monitor level/volume + mix 1/4 and meters, if pressed.

Hm. Although I expected that users can configure digital I/O port (optical interface) and clock configurations (source and frequency), the configuration is just available via GUI software according to manual. (In MOTU 828mkII or later, users can configure them by physical knobs.)

It means that I should update the script so that we can configure these features and check current status. Would you give me a bit time for the additional work?

Thanks for your patience.

danielmellado, 2020-08-06:
Hi @takaswie sure, I have also checked via ffado-mixer, and although the optical I/O can be set to ADAT/Disabled, it doesn't seem to honor it, as well as sample rate. I have been able to change samplerate using jack and can provide another output just so we make sure it works there ;)

```
$ ./motu896hd.py 
Detect clock status:
  0x0b14: 00000018
    rate configuration: 96000
    src configuration: internal
Detect I/O configuration:
  0x0b14: 03000006
    opt out iface configuration: None
    opt in iface configuration: None
```

Regarding the time, for sure! You have been extremely responsive, so thanks a lot! ;)

danielmellado, 2020-08-06:
Ok, so in the end I was able to change them using ffado-mixer and it does seem to work ;), @takaswie that might save you some work ;)

```
[dmellado@fedora  alsa]$ ./motu896hd.py 
Detect clock status:
  0x0b14: 00000000
    rate configuration: 44100
    src configuration: internal
Detect I/O configuration:
  0x0b14: 03000506
    opt out iface configuration: ADAT
    opt in iface configuration: ADAT
```

danielmellado, 2020-08-06:
btw, forgot to mention that I'm also logged in as `dmellado` in freenode's #alsa channel, should you want to ping me. I do also run a bouncer so would appear usually online.

takaswie, 2020-08-25:
Hi @danielmellado ,

I'm sorry to be late for reply, but I was a bit busy for new releases of
libhinawa[1]/libhinoko[2] and their rust bindings[3][4].

> Ok, so in the end I was able to change them using ffado-mixer
> and it does seem to work ;), @takaswie that might save you some work ;)

It might be against your expectation that ALSA firewire stack is not a
port of libffado from user land to kernel land. I have been doing
investigation and reverse-engineering again apart from FFADO
implementation. Thus it is not preferable that the check procedure is
done by FFADO tools, somehow...

The most of MOTU devices have on-board 'cuemix' feature and I can get
current the status of device via LCD by controlling knobs after operating
by software. However, this procedure can not be applied to your case, mmm.

I attempt to get 896HD in used market for reverse-enginnering, but at present there's no guarantee, sorry.

[1] https://github.com/alsa-project/libhinawa
[2] https://github.com/takaswie/libhinoko
[3] https://github.com/alsa-project/hinawa-rs/
[4] hinoko-rs is in my private repository.

takaswie, 2020-08-25:
@danielmellado ,

> I attempt to get 896HD in used market for reverse-enginnering, but at
present there's no guarantee, sorry.

If I fortunately get it, I'll prepare for kernel patch to add support for the device. Before posting, I'm going to ask you for test.

takaswie, 2020-08-27:
> If I fortunately get it, I'll prepare for kernel patch to add support for the device. Before posting, I'm going to ask you for test.

The misfortune comes to me. The bought MOTU 896HD is in disorder and can not detected on IEEE 1394 bus, sigh...

danielmellado, 2020-08-27:
Hi @takaswie, just read all this, wow! Thanks a lot for trying that out! May I help you with some contribution for the MOTU thing? In any case if it's used and not working that's a pain... Should we be leaving somehow close I would ship you mine and lend it to you for the testing... Is there anything I can do from my side?

Basically my idea on the test was, as I can't modify the outputs physically as we'd do with a 828mkII or something, just using the hardware buttons, to do that using the ffado mixer, but only as a way to test that with alsa, so just somehow mimicking the 'hardware buttons'.  Another option would do that I try to hook it up into a amc and try changing the settings with the cuemix (let's see, latest software seems to be "MOTU Universal Audio Installer for Mac OS X (1.6.59644), from 2014".

So, summarizing, if we use ffado-mixer as a way to mimic the settings on the hardware, them later on they seem totally detected from your script, both ADAT and sample rate changes. Is this something we could use from a starting point?

Thanks a lot for your help and efforts!

danielmellado, 2020-08-27:
What we might do is to have a shared tmate session maybe, so you can get access to the hardware, what do you think?

takaswie, 2020-08-30:
Hi,

I'm sorry to be late reply, but was a bit busy for this PR[1].

> Is there anything I can do from my side?
>
> Basically my idea on the test was, as I can't modify the outputs physically as
> we'd do with a 828mkII or something, just using the hardware buttons, to do [...]

Thanks for your dedication and suggestions. Here, I explain my way to work for
ALSA firewire stack.

I have IEEE 1394 protocol hardware analyzer[2]. My work is based on analysis of
actual packet on IEEE 1394 bus, sniffed by the analyzer. I manage to rely on
the analysis result to exclude FFADO implementation from ALSA firewire stack.
This might be against your expectation. But I'm walking on the winding road apart
from FFADO since my work is not porting to ALSA land. (In this point, it's similar
to Wayland project against Xorg, maybe).

So it's our last resort to utilize FFADO stuffs for the work.

> What we might do is to have a shared tmate session maybe, so you can get
> access to the hardware, what do you think?

I open 896HD and ensure that the hardware is quite similar to MOTU Traveler,
expectedly[3]. The hardware consists of four ICs to process isochronous packets
as Traveler does:

 * TI TSB41AB2
 * PDI1394 L40BE
 * Altera Cyclone EP1C3T 144CB
 * TMS320VC5402

I have a conjecture that 896HD can be controlled by the same way for Traveler.
It's mostly reliable that operations in my scripts are appropriate to 896HD.

However, 896HD has some features which Traveler doesn't have. Furthermore,
hardware notification is not clear yet[4].

Anyway, I need time to make enough strategy for 896HD with our cases.

[1] motu: add service program for devices in MOTU FireWire series
https://github.com/alsa-project/snd-firewire-ctl-services/pull/13
[2] DAP Design FireSpy 801
https://www.daptechnology.com/products/bus-analyzer/single-bus/single-bus-gen2/firespy810410bt/
[3] You can see my code comment in source code of Linux kernel.
https://git.kernel.org/pub/scm/linux/kernel/git/tiwai/sound.git/tree/sound/firewire/motu/motu-protocol-v2.c#n173
[4] FFADO doesn't handle the hardware notification.

takaswie, 2021-06-23:
Hi @danielmellado ,

Recently I worked to add support for MOTU 896 in development period for v5.14 kernel[1] and realized that both 896 and 896HD have similar features except for packet streaming function. The experience take me to make inference about specification of 896HD.

My 896HD is still broken. If you are still interested in ALSA support for 896HD, would I ask you
to test two softwares?

1. I pushed `topic/motu-896hd` remote branch, which includes the recent backport drivers and
patches for 896HD. When installing it by DKMS or anything, snd-firewire-motu driver is available
to handle 896HD.

2. I also pushed `topic/motu-896hd` remote branch into snd-firewire-ctl-services project[2]. When
installing, `snd-firewire-motu-ctl-service` program is available via `cargo run`, which can dispatch
requests from ALSA control applications to control 896HD. You can configure 896HD by standard
ALSA control applications such as amixer or alsamixer.

The points we should check are:

* snd-firewire-motu driver can transfer PCM frames/MIDI messages correctly at all of supported
sampling rate. Especially, the type of signal in optical interface is switchable between ADAT and
S/PDIF (and disabled) and packet format changes, thus need to test 3 cases in each of rate.

* snd-firewire-motu-ctl-service program can configure level meters, word clock output speed,
AES/EBU rate converter, the type of signal in optical interface, and source/rate of sampling clock.

I note that CueMix feature is not implemented yet, thus it's preferrable to use phone output,
to which usually the first two channels in PCM frame are assigned.

[1] [PATCH 0/2] ALSA: firewire-motu: add support for MOTU 828 and 896
https://lore.kernel.org/alsa-devel/20210616082847.124688-1-o-takashi@sakamocchi.jp/
[2] https://github.com/alsa-project/snd-firewire-ctl-services

danielmellado, 2021-06-23:
Hi @takaswie, sure! thanks for your efforts on this, I'll take a look as soon as I can allocate some time ;)

takaswie, 2021-06-28:
@danielmellado Thanks to grant my request. As another request, would I ask you to make merge request to add the image of configuration ROM for your MOTU 896HD?

 * https://github.com/takaswie/am-config-roms/

The image is available by below command when detecting the node as `fw1`:

```
$ cat /sys/bus/firewire/devices/fw1/config_rom > am-config-rom/audio_and_music/motu/motu-896.img
```

The content of image is used by Linux FireWire stack to detect operable instances of any node on IEEE 1394 bus. I use the image to create better list of sytemd hardware database:

 * https://github.com/systemd/systemd/blob/main/hwdb.d/80-ieee1394-unit-function.hwdb

For MOTU 896HD, I've already added corresponding entry to the database (see line 981-985). However, metadata in the image is still useful to me.

Regards

takaswie, 2021-07-17:
Hi @danielmellado ,

24 days past since I contact to you but no reactions I have.

If you already lose interests in the topic, I'm sorry to say that I think it better to close the issue. Would I ask your situation?

takaswie, 2021-07-23:
Further 7 days past, but no reaction. I'll close the issue and abandon the topic remote branches.

Thanks

closed 2021-07-23 by takaswie

takaswie, 2021-08-23:
Today I got a new 896HD in used market. Fortunately communication function is alive.

I prepared and posted a path to upstream:
https://lore.kernel.org/alsa-devel/20210823085741.33864-1-o-takashi@sakamocchi.jp/

And it was applied to main branch:
https://git.kernel.org/pub/scm/linux/kernel/git/tiwai/sound.git/commit/?id=23c671be97b9e49846d03ceb0bce3731f4b869ac

The support will be included in Linux kernel v5.15, scheduled to be released a few months later.

danielmellado, 2021-08-23:
Hi Takashi, sorry I didn't have the chance to test this, covid around
family... Thanks for all your help! You made my day! I'll take a look at it!

takaswie, 2021-08-24:
Basic control functions are also available via `snd-firewire-motu-ctl-service` program as well.

[PATCH 0/2] motu-protocols/motu-runtime: add support for MOTU 896HD
https://github.com/alsa-project/snd-firewire-ctl-services/pull/61