takaswie/linux-firewire-dkms #32: Enhacement request: Add "pretty" port names

galileo-pkm, 2020-10-01:
When using JackD it would be beneficial to add port aliases the same way FFado does.
For example, with FFado:

```
[galileo@sol ~]$ jack_lsp -A
firewire_pcm:0040ab0000c39b88_MicIn1 left_in
   system:capture_1
firewire_pcm:0040ab0000c39b88_MicIn1 right_in
   system:capture_2
firewire_pcm:0040ab0000c39b88_LineIn 3+4 left_in
   system:capture_3
firewire_pcm:0040ab0000c39b88_LineIn 3+4 right_in
   system:capture_4
firewire_pcm:0040ab0000c39b88_SpdifIn left_in
   system:capture_5
firewire_pcm:0040ab0000c39b88_SpdifIn right_in
   system:capture_6
firewire_pcm:0040ab0000c39b88_MidiPort_1_in
firewire_pcm:0040ab0000c39b88_LineOut 1+2 left_out
   system:playback_1
firewire_pcm:0040ab0000c39b88_LineOut 1+2 right_out
   system:playback_2
firewire_pcm:0040ab0000c39b88_LineOut 3+4 left_out
   system:playback_3
firewire_pcm:0040ab0000c39b88_LineOut 3+4 right_out
   system:playback_4
firewire_pcm:0040ab0000c39b88_SpdifOut left_out
   system:playback_5
firewire_pcm:0040ab0000c39b88_SpdifOut right_out
   system:playback_6
firewire_pcm:0040ab0000c39b88_MidiPort_1_out

```
with snd_bebob:

```
[galileo@sol ~]$ jack_lsp -A
system:capture_1
   alsa_pcm:hw:FA66:out1
system:capture_2
   alsa_pcm:hw:FA66:out2
system:capture_3
   alsa_pcm:hw:FA66:out3
system:capture_4
   alsa_pcm:hw:FA66:out4
system:capture_5
   alsa_pcm:hw:FA66:out5
system:capture_6
   alsa_pcm:hw:FA66:out6
system:playback_1
   alsa_pcm:hw:FA66:in1
system:monitor_1
system:playback_2
   alsa_pcm:hw:FA66:in2
system:monitor_2
system:playback_3
   alsa_pcm:hw:FA66:in3
system:monitor_3
system:playback_4
   alsa_pcm:hw:FA66:in4
system:monitor_4
system:playback_5
   alsa_pcm:hw:FA66:in5
system:monitor_5
system:playback_6
   alsa_pcm:hw:FA66:in6
system:monitor_6

```
All of my attempts to do so via jack_property have failed ...

takaswie, 2020-10-02:
Hi @galileo-pkm,

Thanks for the idea. I've been considering about how to implement it by ALSA functionalities for a long time.

At the first place, ALSA has no functionality to deliver detail information about each pcm channels into applications. Therefore your idea should involve any change in ALSA core itself.

Actually, you can see the pretty port names for the devices only handled by libffado[1]. On the other hand, you can see no pretty port names for the devices handled by the other backends[2]. It means that you need to encourage ALSA and JACK developers to introduce such functionality into ALSA core, then implement it into JACK code (in Jack::JackAlsaDriver::Attach()).

[1] You can see the corresponding code here: https://github.com/jackaudio/jack2/blob/develop/linux/firewire/JackFFADODriver.cpp#L505
[2] For example, jackd ALSA backend doesn't assign such detail information to the port: https://github.com/jackaudio/jack2/blob/develop/linux/alsa/JackAlsaDriver.cpp#L148

galileo-pkm, 2020-10-02:
Fat chance of that happening. 
I have been testing snd_bebob in last ditch effort
to rule out FFado as the cause of my interface randomly disappearing from the 
firewire bus and the lack of proper channel names is quite annoying. 
Maybe I could nag the Jack developers to add the hint.description from alsa ...

takaswie, 2020-10-03:
> I have been testing snd_bebob in last ditch effort to rule out FFado as the cause of my interface randomly disappearing from the firewire bus

I guess that you encounter the issue reported in the below thread:

https://mailman.alsa-project.org/pipermail/alsa-devel/2020-March/165257.html

FA-66 uses DM1000 chipset produced by BridgeCo. AG with BridgeCo. Enhancement Break Out Box (BeBoB). I know the model sometimes reboot when breaking connection for packet streaming. Unfortunately, I don't know the cause yet.

> Maybe I could nag the Jack developers to add the hint.description from alsa ...

The value of `hint.description` comes from configuration files in alsa-lib. For example:

```
$ cat alsa-lib/src/conf/alsa-.conf
...
pcm.hw {
	...
	type hw
	card $CARD
	device $DEV
	subdevice $SUBDEV
	hint {
		show {
			@func refer
			name defaults.namehint.extended
		}
		description "Direct hardware device without any conversions"
	}
}
```

https://github.com/alsa-project/alsa-lib/blob/master/src/conf/alsa.conf#L181

The field is not designed to give name of each port, against your expectation.

galileo-pkm, 2020-10-03:
I believe that it is a different kind of issue. They report that the device resets.
In my case the device disappears from the bus 
(ffado-test ListDevices does not see it, nothing in the syslog).
Only thing that works is to cycle power on the interface. 

That's with FFado, not your driver and it is really random, happens 1 in 10 times
tricky to diagnose. That's why I switched to snd_bebob but I hit a roadblock: pulseaudio
is acting up while snd_bebob is loaded, breaks in the audio, locks up after a while. 
I will test further, if I get some free time later and post those logs if you are interested in the issue.

closed 2025-08-21 by galileo-pkm