takaswie/linux-firewire-dkms #33: snd-dice / TC Electronic Konnekt 24D

capoei, 2020-10-31:
snd-dice never worked for my device. I guess the issue is known? the sound is crackling, and these are no x-runs. It was ok for me all those years since ffado works great for this device. now I am interested in snd-dice again whatever, since I would like to use Pipewire which doesn't suport ffado. If there is anything I can do to help debug the issue let me know

takaswie, 2020-10-31:
Hi @capoei,

I'm sorry for the inconvenience. I use all of models produced by TC Electronic with ASICs of DICE series to test ALSA dice driver:
 - Konnekt 24D (DiceII)
 - Konnekt 8 (DiceII)
 - Konnekt Live (DiceII)
 - Impact Twin (DiceII)
 - Studio Konnekt 48 (DiceII + TCD2210)
 - Digital Konnekt x32 (DiceII + TCD2210)
 - Desktop Konnekt 6 (TCD2220)

All of them are a kind of device which uses unique sampling transfer frequency and expects drivers to recover media transfer clock on incoming packet for outgoing packet. In short, the device doesn't work with fidelity sampling rate, at least on the series of packet on IEEE 1394 bus, even if configured for well-known sampling rate such as 48.0 kHz.

At present, ALSA IEC 61883-1/6 packet streaming engine is designed to use sampling transfer frequency exactly as the same as sampling rate configured by ALSA PCM applications, ignoring the media transfer clock. This can cause the kind of device to generate sound with noises.

I've been working to improve the packet streaming engine this year but not finished yet. For the work, developer needs enough knowledge about IEC 61883-1/6 packet streaming and quirks of actual devices retrieved by appropriate way without any biases. Therefore at present I don't expect any support from user unless the user have enough knowledge.

For your information, in another project to implement control service in userspace, I did reverse-engineering for the models of TC Electronic and clear the way to control them somehow. I publish Google Spreadsheet for register mapping[2] and Rust implementation in `topic/dice` remote branch of the repository.

Thanks

[1] https://github.com/alsa-project/snd-firewire-ctl-services/
[2] https://docs.google.com/spreadsheets/d/1hb86pABzKT8abvgVVQoObUQEAMav6bQ05jA6yVwNHaY/edit#gid=0

capoei, 2020-11-04:
Hi @takaswie what a pitty.
is there a updated list of win/mac-obsolete great pro devices that work flawlessly with the alsa driver?

takaswie, 2020-11-04:
> is there a updated list of win/mac-obsolete great pro devices that work flawlessly with the alsa driver?

The meaning of `pro` differs depending on context (e.g. audio listening, audio recording, audio mastering and so on). It's better to describe your needs when posting such kind of question as detal as possible, in my opinion.

capoei, 2020-11-05:
> The meaning of `pro` differs depending on context (e.g. audio listening, audio recording, audio mastering and so on). It's better to describe your needs when posting such kind of question as detal [...]

I am thinking of something like Presonus Firestudio Project or Digi 002 Rack for example. both which I saw used for cheap only to find out that their alsa drivers also have problems

takaswie, 2020-11-05:
> I am thinking of something like Presonus Firestudio Project or Digi 002 Rack for example. both which I saw used for cheap only to find out that their alsa drivers also have problems

Presonus Firestudio Project uses TCD2210 and support TCAT protocol extension. Digidesign Digi002 rack uses FPGA[1] to process payload of isochronous packet by own way.

Both models use different way to transfer media transfer clock, however in a point of expecting drivers to recover the media transfer clock on incoming packet for outgoing packet, they are the same.

[1] https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/sound/firewire/digi00x?id=9edf723fd85822c7b7d8ef4f41a74c5a33eeca0c

capoei, 2020-11-06:
well, both have problems with the driver, right?

takaswie, 2021-04-16:
Hi @capoei ,

Just now I pushed patches to `topic/media-clock-recovery` remote branch (f3bdd2c269d1).
Would I request your cooperation to debug the patches?

With this patchset, ALSA IEC 61883-1/6 packet streaming engine processes incoming and
outgoing packets by:

1. Transfer outgoing packet with pre-computed table rate in its beginning.
2. Pool presentation timestamp in incoming packets after starting receiving them
3. Then transfer outgoing packet by media clock recovered from the presentation timestamp

In a prompt test, it looks well with my TC Electronic Konnekt 24d at all of supported sampling
rate (44.1/48.0/88.2/96.0/1764.0/192.0 kHz). I wish it also works well in your side.

Cheers.

capoei, 2021-04-16:
> Hi @capoei ,
> 
> Just now I pushed patches to `topic/media-clock-recovery` remote branch ([f3bdd2c](https://github.com/takaswie/snd-firewire-improve/commit/f3bdd2c269d189d7ff9652a919f77ee52316ef36)).
> Would I request your cooperation to debug the patches?
> 
> With this patchset, ALSA IEC 61883-1/6 packet streaming engine processes incoming and
> outgoing packets by:
> 
> 1. Transfer outgoing packet with pre-computed table rate in its beginning.
> 2. Pool presentation timestamp in incoming packets after starting receiving them
> 3. Then transfer outgoing packet by media clock recovered from the presentation timestamp
> 
> In a prompt test, it looks well with my TC Electronic Konnekt 24d at all of supported sampling
> rate (44.1/48.0/88.2/96.0/1764.0/192.0 kHz). I wish it also works well in your side.
> 
> Cheers.

Hi @takaswie 
surprised you are working on this, very cool.
I must say that I am a long time Arch user who is used to using AUR for aything which is not in the repos. I have no clue how to install this and aply patches (lol). with that beeing said, I am very intrested in the subject and will try to get my head around it soon. I´ll let you know

alesandar, 2021-04-18:
@takaswie, I decided to take a look at topic/media-clock-recovery and I can see that you've implemented some interesting stuff in sound/firewire/amdtp-stream.c, but your last commit (f3bdd2c269d189d7ff9652a919f77ee52316ef36) affects only the dice module. I am not sure if this will actually affect the corresponding modules for Motu 828mk3 and Digi 002 Rack without any further changes. I know how to use DKMS and I am willing to test your recent changes with both devices.

takaswie, 2021-04-18:
Hi @alesandar,

> I decided to take a look at topic/media-clock-recovery and I can see that you've implemented some interesting stuff in sound/firewire/amdtp-stream.c, but your last commit (f3bdd2c) affects only the dice module. I am not sure if this will actually affect the corresponding modules for Motu 828mk3 and Digi 002 Rack without any further changes. I know how to use DKMS and I am willing to test your recent changes with both devices.

If you have interests in the patches, I feel happy. However, they are not applied to the devices as is since the devices uses complete different mechanism for media clock recovery.

Within devices supported by drivers in ALSA firewire stack, the level of Dice devices to be complianant to IEC 61883-1/6 is the best. On the other hand, devices in MOTU firewire series uses own way. The devices use timestamp for media clock recovery, however the way is not standard defined in IEC 61883-1/6. Furthermore, devices in Digi 00x series doesn't use timestamp itself for media clock recovery.

Like this, each category of device require specific implementation for media clock recovery. This is the first step for me to integrate toward it...

takaswie, 2021-06-01:
I note that patchset for the issue is posted to upstream:

* https://lore.kernel.org/alsa-devel/20210601081753.9191-1-o-takashi@sakamocchi.jp/

takaswie, 2021-06-01:
The patchset is merged to upstream. The patches are also available in master branch of the repository.

The service program to control the device is available. Please refer to below URL if you are interested:

* https://github.com/alsa-project/snd-firewire-ctl-services/

Let me close the issue. Thanks for your cooperation.

closed 2021-06-01 by takaswie