takaswie/linux-firewire-dkms #36: M-Audio ProFire Lightbridge: Kernel panic

jpnielsen, 2021-07-21:
Trying to revive old "M-Audio ProFire LightBridge" interface.

It's attached using adat to one PreSonus Digimax FS (I also have a Behringer Ultragain Pro-8 and an Expert Sleepers ES-3 with adat)
no-name Firewire pci-e interface (VIA VT6307 + asmedia asm1083 ICs)
motherboard is an ASRock Phantom Gaming 4 / AMD X570
CPU is a Ryzen 7 3700x
RAM is 2x16GB Corsair CMK32GX4M2B3200C16

uname -a
Linux debian 5.10.0-8-amd64 #1 SMP Debian 5.10.46-1 (2021-06-24) x86_64 GNU/Linux

lspci |grep -i firewire
05:00.0 FireWire (IEEE 1394): VIA Technologies, Inc. VT6306/7/8 [Fire II(M)] IEEE 1394 OHCI Controller (rev 80)

I get a Kernel Panic when I turn on the "M-Audio ProFire LightBridge".

[ 96.930836] mce: [Hardware Error]: CPU 6: Machine Check Exception: 7 Bank 0: bc002800000c0135
[ 96.930837] mce: [Hardware Error]: RIP 10:ffffffffc025b08a> {get_cycle_time+0x2a/0x120 [firewire_ohci]}
[ 96.930838] mce: [Hardware Error]: TSC 69e4ebbb40 ADDR fc7000c0 MISC d01a0ffe00000000 IPID b000000000
[ 96.930840] mce: [Hardware Error]: PROCESSOR 2:870f10 TIME 1626800481 SOCKET 0 APIC c microcode 8701021
[ 96.930840] Kernel panic - not syncing: Fatal local machine check
[ 96.931077] Kernel Offset: 0x2b000000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)

The "M-Audio ProFire LightBridge" using the M-Audio drivers for windows 7 SP1 (released 06/02/2012):
Using the M-Audio system-tray tool, I can set the sample rate and sync and the LED's change accordingly on the ProFire.
But when i try sending audio to it, windows crashes.
https://m-audio.com/support/download/drivers/firewire-installer-6.0.4-driver-5.10.0.5061

I don't have other firewire devices for testing the VIA firewire adapter.

Am I missing something obvious?
Any tips on how to proceed ?

Thank you for your contributions to the community.

jpnielsen, 2021-07-21:
Perhaps of use, i installed Ubuntu Studio 21.04

```
uname -a

Linux ubuntustudio 5.11.0-25-lowlatency #27-Ubuntu SMP PREEMPT Sat Jul 10 00:06:29 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Ubuntustudio 21.04, Booted to KDE, Switched to tty with Ctrl-Shift-F3, turned on ProFire Lightbridge - when it stops booting, a panic (took picture and OCR+edited):
 
[  268.951611] mce: [Hardware Error]: CPU 13: Machine Check Exception: 7 Bank 0: bc002800000c0135 
[  268.951614] mce: [Hardware Error]: RIP 10:<ffffffffc05b61f0> {getcycletime+0x10/0x120 [firewire_ohci]} 
[  268.351616] mce: [Hardware Error]: TSC ffc231bc28 ADOR fc7000c0 MISC d01a000000000000 IPID b000000000 
[  268.351618] mce: [Hardware Error]: PROCESSOR 2:870f10 TIME 1626843309 SOCKET 0 APIC b microcode 8701021 
[  268.551619] Kernel panic - not syncing: Fatal local machine check
[  268.952114] Kernel Offset: 0x9400000 from Oxffffffff81000000 (relocatiom range: 0xffffffff80000000-0xffffffffbfffffff)

grep firewire /var/log/syslog

Jul 21 06:10:45 ubuntustudio kernel: [    0.658424] firewire_ohci 0000:05:00.0: enabling device (0080 -> 0083)
Jul 21 06:10:45 ubuntustudio kernel: [    0.714658] firewire_ohci 0000:05:00.0: added OHCI v1.10 device as card 0, 4 IR + 8 IT contexts, quirks 0x11
Jul 21 06:10:45 ubuntustudio kernel: [   13.913262] firewire_core 0000:05:00.0: created device fw0: GUID 0011223333666677, S400
```

jpnielsen, 2021-07-21:
Let me add the version number found in the Windows software (version 6.0.4) for the M-Audio ProFire LightBridge:
```
bootloader: 2006/03/14 17:51:18
firmware: 2006/09/14 14:54:30
hardware: model:8 version: 0
GUID: 000D-6C08-00A6-7063
```

jpnielsen, 2021-07-22:
News - New kernel - no panic.

But, pulseaudiocontrol takes a long time to come up - and the Firewire<>Adat box is not seen.

Behringer Ultragain Pro-8, adat sync blinks irregularly
Presonus Digimax FS set to adat sync locks, blinks, locks ...
The adat devices are set to sync to adat

M-Audio Profire Lightbridge LEDs:
44.1k is lit
Sync source Int is blinking irregularly.

```

uname -a
Linux manjaro 5.14.0-rc2-next-20210722custom #1 SMP PREEMPT Thu Jul 22 18:08:41 CEST 2021 x86_64 GNU/Linux

dmesg

[  137.815364] firewire_ohci 0000:05:00.0: isochronous cycle inconsistent
[  138.329420] firewire_core 0000:05:00.0: created device fw1: GUID 000d6c0800a6706e, S400
[  138.329611] firewire_core 0000:05:00.0: phy config: new root=ffc1, gap_count=5
[  197.749370] snd_bebob fw1.0: FCP command timed out
[  197.749377] snd_bebob fw1.0: fail to get section type for isoc out plug 0: -5
[  198.133359] snd_bebob fw1.0: FCP command timed out
[  198.435260] snd_bebob fw1.0: transaction failed: conflict error
[  198.481281] snd_bebob fw1.0: transaction failed: conflict error
[  198.526292] snd_bebob fw1.0: transaction failed: conflict error
[  198.571278] snd_bebob fw1.0: transaction failed: conflict error
[  198.571286] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.616471] snd_bebob fw1.0: transaction failed: conflict error
[  198.616478] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.662898] snd_bebob fw1.0: transaction failed: conflict error
[  198.662906] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.708284] snd_bebob fw1.0: transaction failed: conflict error
[  198.708291] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.760286] snd_bebob fw1.0: transaction failed: conflict error
[  198.760293] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.805274] snd_bebob fw1.0: transaction failed: conflict error
[  198.805282] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.849471] snd_bebob fw1.0: transaction failed: conflict error
[  198.849478] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.893531] snd_bebob fw1.0: transaction failed: conflict error
[  198.893539] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.937463] snd_bebob fw1.0: transaction failed: conflict error
[  198.937470] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.981517] snd_bebob fw1.0: transaction failed: conflict error
[  198.981525] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  393.604694] snd_bebob fw1.0: Detect discontinuity of cycle: 58224 62238
[  418.105681] snd_bebob fw1.0: Detect discontinuity of cycle: 62226 2240
[  458.114359] snd_bebob fw1.0: Detect discontinuity of cycle: 62238 2253
[  536.859305] snd_bebob fw1.0: Detect discontinuity of CIP: C0 C8
[  593.064260] snd_bebob fw1.0: Detect discontinuity of CIP: B8 C0
[  618.125590] snd_bebob fw1.0: Detect discontinuity of cycle: 62256 2270
[  672.872016] snd_bebob fw1.0: Detect discontinuity of CIP: B8 C0
[  713.031817] snd_bebob fw1.0: Detect discontinuity of CIP: B0 B8
[  738.118130] snd_bebob fw1.0: Detect discontinuity of cycle: 62285 2299
[  864.896532] snd_bebob fw1.0: Detect discontinuity of CIP: B0 B8
[  953.089148] snd_bebob fw1.0: Detect discontinuity of CIP: B0 B8
[  992.920674] snd_bebob fw1.0: Detect discontinuity of CIP: C0 C8
[ 1273.103460] snd_bebob fw1.0: Detect discontinuity of CIP: A8 B0
[ 1313.015606] snd_bebob fw1.0: Detect discontinuity of CIP: B0 B8
[ 1338.166550] snd_bebob fw1.0: Detect discontinuity of cycle: 62461 2476

```

takaswie, 2021-07-22:
Hi,

> News - New kernel - no panic.

That sounds good. As long as I know, many users of Ryzen Zen 2 SoC encounter the similar issue. I guess it comes from the other part of functions irrelevant to IEEE 1394 bus and M-Audio ProFire Lightbridge.

Anyway the kernel panic issue is fixed in your side. Let you close the issue and file another one for the discontinuous issue.

jpnielsen, 2021-07-22:
pulseaudiocontrol takes a long time to come up - and the Firewire<>Adat box is not seen.

Behringer Ultragain Pro-8, adat sync blinks irregularly
Presonus Digimax FS set to adat sync locks, blinks, locks ...
The adat devices are set to sync to adat

M-Audio Profire Lightbridge LEDs:
44.1k is lit
Sync source Int is blinking irregularly.
```

uname -a
Linux manjaro 5.14.0-rc2-next-20210722custom #1 SMP PREEMPT Thu Jul 22 18:08:41 CEST 2021 x86_64 GNU/Linux

dmesg

[  137.815364] firewire_ohci 0000:05:00.0: isochronous cycle inconsistent
[  138.329420] firewire_core 0000:05:00.0: created device fw1: GUID 000d6c0800a6706e, S400
[  138.329611] firewire_core 0000:05:00.0: phy config: new root=ffc1, gap_count=5
[  197.749370] snd_bebob fw1.0: FCP command timed out
[  197.749377] snd_bebob fw1.0: fail to get section type for isoc out plug 0: -5
[  198.133359] snd_bebob fw1.0: FCP command timed out
[  198.435260] snd_bebob fw1.0: transaction failed: conflict error
[  198.481281] snd_bebob fw1.0: transaction failed: conflict error
[  198.526292] snd_bebob fw1.0: transaction failed: conflict error
[  198.571278] snd_bebob fw1.0: transaction failed: conflict error
[  198.571286] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.616471] snd_bebob fw1.0: transaction failed: conflict error
[  198.616478] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.662898] snd_bebob fw1.0: transaction failed: conflict error
[  198.662906] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.708284] snd_bebob fw1.0: transaction failed: conflict error
[  198.708291] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.760286] snd_bebob fw1.0: transaction failed: conflict error
[  198.760293] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.805274] snd_bebob fw1.0: transaction failed: conflict error
[  198.805282] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.849471] snd_bebob fw1.0: transaction failed: conflict error
[  198.849478] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.893531] snd_bebob fw1.0: transaction failed: conflict error
[  198.893539] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.937463] snd_bebob fw1.0: transaction failed: conflict error
[  198.937470] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  198.981517] snd_bebob fw1.0: transaction failed: conflict error
[  198.981525] snd_bebob fw1.0: fail to get an input for MSU in plug 7: -5
[  393.604694] snd_bebob fw1.0: Detect discontinuity of cycle: 58224 62238
[  418.105681] snd_bebob fw1.0: Detect discontinuity of cycle: 62226 2240
[  458.114359] snd_bebob fw1.0: Detect discontinuity of cycle: 62238 2253
[  536.859305] snd_bebob fw1.0: Detect discontinuity of CIP: C0 C8
[  593.064260] snd_bebob fw1.0: Detect discontinuity of CIP: B8 C0
[  618.125590] snd_bebob fw1.0: Detect discontinuity of cycle: 62256 2270
[  672.872016] snd_bebob fw1.0: Detect discontinuity of CIP: B8 C0
[  713.031817] snd_bebob fw1.0: Detect discontinuity of CIP: B0 B8
[  738.118130] snd_bebob fw1.0: Detect discontinuity of cycle: 62285 2299
[  864.896532] snd_bebob fw1.0: Detect discontinuity of CIP: B0 B8
[  953.089148] snd_bebob fw1.0: Detect discontinuity of CIP: B0 B8
[  992.920674] snd_bebob fw1.0: Detect discontinuity of CIP: C0 C8
[ 1273.103460] snd_bebob fw1.0: Detect discontinuity of CIP: A8 B0
[ 1313.015606] snd_bebob fw1.0: Detect discontinuity of CIP: B0 B8
[ 1338.166550] snd_bebob fw1.0: Detect discontinuity of cycle: 62461 2476
```

closed 2021-07-22 by jpnielsen

jpnielsen, 2021-07-22:
I did buy a 5m FireWire cable - the FW wikipedia page says 4.5m max - perhaps that is part of the explanation for the errrors ?