takaswie/linux-firewire-dkms #37: M-Audio ProFire Lightbridge: Discontinous errors

jpnielsen, 2021-07-22:
Hi,

> News - New kernel - no panic.

That sounds good. As long as I know, many users of Ryzen Zen 2 SoC encounter the similar issue. I guess it comes from the other part of functions irrelevant to IEEE 1394 bus and M-Audio ProFire Lightbridge.

Anyway the kernel panic issue is fixed in your side. Let you close the issue and file another one for the discontinuous issue.

_Originally posted by @takaswie in https://github.com/takaswie/snd-firewire-improve/issues/36#issuecomment-885298235_

takaswie, 2021-07-23:
Thanks for the report.

I have M-Audio ProFire Lightbridge as one of my test device, and have realized that the
device has some quirks in its packet streaming. As a result, ALSA bebob driver often detects
packet discontinuity and cancels data transmission, as you can see in your kernel log:

```
[  458.114359] snd_bebob fw1.0: Detect discontinuity of cycle: 62238 2253
[  536.859305] snd_bebob fw1.0: Detect discontinuity of CIP: C0 C8
```

The first line means that the device stops packet transmission during 4015 isochronous
cycles. The second line means that the device doesn't multiplex PCM frames in an
isochronous packet.

You can see my comment in patch for Linux kernel as well:

 * commit 5ec85c198eff ("ALSA: bebob: distinguish M-Audio ProFire Lightbridge quirk")
    * https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/sound/firewire/bebob?id=5ec85c198eff077e99bc5ff615eb9618004962e2

Inconvenient to you, I have not found remedy against it yet. This means that at present it's
a bit hard to use the device with PulseAudio or PipeWire since the sound servers manage
to start/stop data transmission several times to retrieve parameters of PCM substream
when detecting new device.

If saying that we have possibility to use the device, it might be with jackd in Jack Audio
Connection Kit since it doesn't have function to detect new device automatically. In the
case, uses need to try starting the service several times till the device finally transfers
stable sequence of packet.

I'm sorry to trouble you but it's the best what we can do at present...

jpnielsen, 2021-07-23:
Thank you.

I have in the meantime got the M-Audio ProFire LightBridge interface working in Windows 7 SP1 (32bit) 
- on another computer - a HP Pavillion p6000. Intel core i3. 8GB ram. Nvidia geforce 610.  - with VIA FireWire onboard controller.

My best wishes
Jens Peter Nielsen

takaswie, 2021-07-23:
I'd like to fix the issue, however at present I have no idea. I keep the issue opened to tracking future improvements of ALSA firewir stack for the device.

jpnielsen, 2021-07-23:
new test on windows 10 20H1 = Everything works.

HP Pavillion p6000. Intel core i3. 8GB ram. Nvidia geforce 610. - with the VIA FireWire PCI-E card I used in the AMD Zen 2 PC before.

I will now update to Windows 10 21H1 and see if it makes it crash like on the other PC before.
And after that install the same Linux that gave the errors on the other PC.

jpnielsen, 2021-07-24:
HP Pavillion p6000. Intel core i3. 8GB ram. Nvidia geforce 610. - with VIA FireWire onboard controller.

Everything works in Windows 10 19043.928

jpnielsen, 2021-08-08:
An update.

I received a new PCI-E firewire card, It works on the PC that didn't work with the VIA FW card before. Below are the errors it gives in Linux. The Presonus ADA converter fails to lock sync to the M-Audio Profire Lightbridge.

Linux manjaro 5.14.0-rc2-next-20210722custom #1 SMP PREEMPT Thu Jul 22 18:08:41 CEST 2021 x86_64 GNU/Linux

lspci | grep -i firewire

06:00.0 FireWire (IEEE 1394): Texas Instruments TSB82AA2 IEEE-1394b Link Layer Controller (rev 01)

[  307.250565] firewire_core 0000:06:00.0: phy config: new root=ffc1, gap_count=5
[  307.538461] firewire_core 0000:06:00.0: phy config: new root=ffc1, gap_count=5
[  307.768216] firewire_core 0000:06:00.0: created device fw1: GUID 000d6c0800a6706e, S400
[  340.057987] snd_bebob fw1.0: Detect discontinuity of CIP: B0 B8
[  371.626481] snd_bebob fw1.0: Detect discontinuity of CIP: B8 C0
[  420.058488] snd_bebob fw1.0: Detect discontinuity of CIP: B0 B8
[  444.559352] snd_bebob fw1.0: Detect discontinuity of CIP: B8 C0
[  467.642537] snd_bebob fw1.0: Detect discontinuity of CIP: C0 C8
[  540.089429] snd_bebob fw1.0: Detect discontinuity of CIP: B0 B8

 aplay -l
**** List of PLAYBACK Hardware Devices ****
card 1: UMC1820 [UMC1820], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Audio [Rift Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: Lightbridge [ProFire Lightbridge], device 0: BeBoB [ProFire Lightbridge PCM]
  Subdevices: 1/1
  Subdevice #0: subdevice #0