takaswie/linux-firewire-dkms #38: [Mackie Onyx 1640i] Kernel module snd-oxfw plays audio in slow motion.

AreYouLoco, 2021-09-11:
As I already wrote an email to you maybe some other developers might want to take a look so I re-post here:

So I confirm my Mackie Onyx 1640i is using snd_oxfw driver.

> I have two items to request you:
> 
>  * retrieve packet dump from ALSA oxfw driver
>  * test with the latest driver

1. Here is dump [1] (file:trace.dat_stock) with the old (stock) driver for my custom kernel (5.12.2-rt3-fw) and the [1] (file:trace.dat_new) from your new driver (master branch) that you asked to test out. Everything at 512 buffer size and 3 periods @ 48kHz and JACK2 with ALSA backend (44,1kHz test described below) :

The problem is when I play music from pulseaudio sink (from Firefox; Soundcloud and YouTube) the audio plays in slow motion on both sites mentioned. Even the video on YouTube is in slow motion. With VLC and local mp3 file there is slow motion effect and interrupt/break in audio output every 1 second. In Ardour (Linux DAW) the same problem appears. Audio is slowed or just really pitched down. I used command `sleep 5` to be more on less precise on the time `sudo trace-cmd record -e snd_firewire_lib:amdtp_packet` runs.

Also from my experience; I played audio only for short time on YT (pulseaudio_sink to ALSA backend on JACK as I dont have 44.1kHz sessions in Ardour. But at 44.1kHz the same slow motion effect appears on YT and Soundcloud as well as files played locally via VLC. (file:trace.dat_441)

The only one message from dmesg is:
`snd_oxfw fw1.0: Detect discontinuity of CIP: 68 00`

But I am not sure when it appeared. With old or the new master kernel module. Maybe I will try to figure it out.

LINK TO THE DUMPS
[1] https://mega.nz/folder/76hCBRTK#muOYoyghB8fbva4alQDZnQ
    
I've verified I run your new driver with this:
```
areyouloco@studio:~/code/snd-firewire-improve$ sudo modinfo snd-oxfw
filename:       /lib/modules/5.12.2-rt3-fw/updates/dkms/snd-oxfw.ko
sudo dkms install alsa-firewire/4.17
Module alsa-firewire/4.17 already installed on kernel 5.12.2-rt3-fw/x86_64
```
(sudo modprobe -r before and loaded the module again after DKMS build to be sure)

2. So when I uninstall your driver and run it from kernel directory:
```
areyouloco@studio:~/code/snd-firewire-improve$ sudo modinfo snd-oxfw
filename: /lib/modules/5.12.2-rt3-fw/kernel/sound/firewire/oxfw/snd-oxfw.ko

```
(again sudo modprobe -r before and loaded the module again after DKMS uninstall)

After restoring to kernel's stock module it plays at normal speed/pitch.

So there is for sure some bug for sure to be rechecked.

Thank you for your amazing work! Up to some further tests,
AreYouLoco?

AreYouLoco, 2021-09-11:
I've rechecked my dmesg output and found some related lines:
```
[    6.962065] firewire_ohci 0000:0d:00.3: added OHCI v1.10 device as card 0, 4 IR + 4 IT contexts, quirks 0x50
[    6.962190] firewire_ohci 0000:0d:00.3: isochronous cycle inconsistent
[    6.962313] firewire_ohci 0000:0d:00.3: isochronous cycle too long
[    6.962437] firewire_ohci 0000:0d:00.3: isochronous cycle too long
[    6.962560] firewire_ohci 0000:0d:00.3: isochronous cycle too long
[    6.962685] firewire_ohci 0000:0d:00.3: isochronous cycle too long
[    6.962813] firewire_ohci 0000:0d:00.3: isochronous cycle too long
[    6.962936] firewire_ohci 0000:0d:00.3: isochronous cycle too long
[    6.963060] firewire_ohci 0000:0d:00.3: isochronous cycle too long
[    6.963185] firewire_ohci 0000:0d:00.3: isochronous cycle too long
[    6.963313] firewire_ohci 0000:0d:00.3: isochronous cycle too long
[    7.466963] firewire_core 0000:0d:00.3: created device fw0: GUID 0021cc20090c29ce, S400
[    7.474158] firewire_core 0000:0d:00.3: created device fw1: GUID 000ff2050000145b, S400
[    7.474162] firewire_core 0000:0d:00.3: phy config: new root=ffc0, gap_count=5
[ 6759.166861] firewire_core 0000:0d:00.3: rediscovered device fw0
```
Not sure using which driver. Will try to establish which driver correlates to that messages.

takaswie, 2021-09-11:
Thanks for the report! As a quick glance of the packet dump, it comes from the quirk specific to your model or mackie models. I'd like you to wait a bit till I finish further analysis.

Cheers.

takaswie, 2021-09-11:
@AreYouLoco Would I request you to ssend additional information below?

* dump of/proc/asound/cards includes values ofsome registers.

Thanks

AreYouLoco, 2021-09-11:
> @AreYouLoco Would I request you to ssend additional information below?
>     * dump of/proc/asound/cards includes values ofsome registers.
> Thanks

```
areyouloco@studio:~$ cat /proc/asound/cards
 0 [PCH            ]: HDA-Intel - HDA Intel PCH
                      HDA Intel PCH at 0xf3820000 irq 34
 1 [NVidia         ]: HDA-Intel - HDA NVidia
                      HDA NVidia at 0xf1000000 irq 17
 2 [O1640i         ]: OXFW - Onyx 1640i
                      Loud Technologies Inc. Onyx 1640i (OXFW971 0105), GUID 000ff2050000145b at fw1.
```

takaswie, 2021-09-12:
I hastily wrote Python 3 script to parse the packet dumps.

```python
#!/usr/bin/env python3

from sys import argv
from pathlib import Path

def build_quadlet(elems):
    return (int(elems[0]) << 24) | \
           (int(elems[1]) << 16) | \
           (int(elems[2]) << 8) | \
           (int(elems[3]) << 0)

if len(argv) < 2:
    print('At least one argument is required for path of file')
    sys.exit(1)
filepath = argv[1]

path = Path(filepath)
if not path.exists():
    print('File not found')
    sys.exit(1)

seq = {}

with path.open('r') as f:
    for line in f:
        line = line.strip();
        pos = line.find('amdtp_packet')
        if pos < 0:
            continue
        elems = line[(pos + 22):].split(' ')

        sec = int(elems[0])
        cycle = int(elems[1])
        src_node = elems[2]
        dst_node = elems[3]
        isoc_ch = int(elems[4])
        quadlet_count = int(elems[5])
        data_blocks = int(elems[6])
        data_block_count = int(elems[7])
        packet_index = int(elems[8])
        is_irq = int(elems[9]) > 0
        index = int(elems[10])

        cip0 = build_quadlet(elems[-8:-4])
        cip1 = build_quadlet(elems[-4:])

        if isoc_ch not in seq:
            seq[isoc_ch] = []
        seq[isoc_ch].append([sec, cycle, quadlet_count, data_blocks, data_block_count, packet_index, is_irq, index, cip0, cip1])

def compute_cycle_count(sec, cycle):
    return (sec * 8000) + cycle

def increment_cycle_count(cycle_count):
    cycle_count += 1
    return cycle_count % (8 * 8000)

for isoc_ch, cycles in seq.items():
    next_cycle_count = compute_cycle_count(cycles[0][0], cycles[0][1])
    count = 0
    events = 0;
    
    print('ch {}:'.format(isoc_ch))
    for cycle in cycles:
        cycle_count = compute_cycle_count(cycle[0], cycle[1])
        if cycle_count != next_cycle_count:
            print('  discontinue')
        next_cycle_count = increment_cycle_count(cycle_count)

        events += cycle[3]

        count += 1
        if count == 8000:
            print('  {} sec, {} cycle: {} events'.format(cycle[0], cycle[1], events))
            count = 0
            events = 0
```

When parsing, the packet dump should be converted for text format by `trace-cmd report`.
As results:

```
$ ./parse stock.log
ch 0:
  1 sec, 5594 cycle: 48000 events
  2 sec, 5594 cycle: 48000 events
  3 sec, 5594 cycle: 48000 events
  4 sec, 5594 cycle: 48000 events
  5 sec, 5594 cycle: 48000 events
  6 sec, 5594 cycle: 48000 events
  7 sec, 5594 cycle: 48000 events
  0 sec, 5594 cycle: 48000 events
  1 sec, 5594 cycle: 48000 events
  2 sec, 5594 cycle: 48000 events
  3 sec, 5594 cycle: 48000 events
  4 sec, 5594 cycle: 48000 events
ch 1:
  1 sec, 5336 cycle: 48000 events
  2 sec, 5336 cycle: 48000 events
  3 sec, 5336 cycle: 48000 events
  4 sec, 5336 cycle: 48000 events
  5 sec, 5336 cycle: 48000 events
  6 sec, 5336 cycle: 48000 events
  7 sec, 5336 cycle: 48000 events
  0 sec, 5336 cycle: 48000 events
  1 sec, 5336 cycle: 48000 events
  2 sec, 5336 cycle: 48000 events
  3 sec, 5336 cycle: 48000 events
  4 sec, 5336 cycle: 48000 events
```

```
$ ./parse new.log
ch 0:
  5 sec, 4318 cycle: 26392 events
  6 sec, 4318 cycle: 26400 events
  7 sec, 4318 cycle: 26392 events
  0 sec, 4318 cycle: 26392 events
  1 sec, 4318 cycle: 26400 events
  2 sec, 4318 cycle: 26392 events
  3 sec, 4318 cycle: 26392 events
  4 sec, 4318 cycle: 26400 events
  5 sec, 4318 cycle: 26392 events
ch 1:
  5 sec, 4037 cycle: 26400 events
  6 sec, 4037 cycle: 26392 events
  7 sec, 4037 cycle: 26400 events
  0 sec, 4037 cycle: 26392 events
  1 sec, 4037 cycle: 26392 events
  2 sec, 4037 cycle: 26400 events
  3 sec, 4037 cycle: 26392 events
  4 sec, 4037 cycle: 26392 events
  5 sec, 4037 cycle: 26400 events
```

```
$ ./parse 441.log
ch 0:
  0 sec, 3825 cycle: 26336 events
  1 sec, 3825 cycle: 26336 events
  2 sec, 3825 cycle: 26336 events
  3 sec, 3825 cycle: 26336 events
  4 sec, 3825 cycle: 26336 events
  5 sec, 3825 cycle: 26336 events
  6 sec, 3825 cycle: 26336 events
  7 sec, 3825 cycle: 26336 events
  0 sec, 3825 cycle: 26336 events
  1 sec, 3825 cycle: 26336 events
  2 sec, 3825 cycle: 26344 events
  3 sec, 3825 cycle: 26336 events
  4 sec, 3825 cycle: 26336 events
  5 sec, 3825 cycle: 26336 events
ch 1:
  0 sec, 3567 cycle: 26336 events
  1 sec, 3567 cycle: 26336 events
  2 sec, 3567 cycle: 26336 events
  3 sec, 3567 cycle: 26336 events
  4 sec, 3567 cycle: 26336 events
  5 sec, 3567 cycle: 26336 events
  6 sec, 3567 cycle: 26336 events
  7 sec, 3567 cycle: 26336 events
  0 sec, 3567 cycle: 26336 events
  1 sec, 3567 cycle: 26336 events
  2 sec, 3567 cycle: 26344 events
  3 sec, 3567 cycle: 26336 events
  4 sec, 3567 cycle: 26336 events
  5 sec, 3567 cycle: 26336 events
```

In all cases, the device uses isochronous channel number 1. We can see apparent regression
between v5.14 kernel and v5.13 kernel or before. 55 % of packets than media clock frequency are actually transferred. It is the cause of slow playback.

Fortunately, at least, we know the way to handle the device correctly ;) I prepare a patch and push it into `topic/fix-oxfw-mackie-regression` remote branch. Would I ask you to test it in your system?
Regardless of success or failure, It's preferable to save packet dump and share it to me.

Thanks

AreYouLoco, 2021-09-12:
```
areyouloco@studio:~/code/snd-firewire-improve$ git status
HEAD detached at origin/topic/fix-oxfw-mackie-regression
nothing to commit, working tree clean
```

Audio and video is still in slow motion. Same result in VLC. I can see no real difference with this fix.
Packet dump: https://mega.nz/file/figBmCga#NXsEfFFprKNt552jEbjtMrlFXsbrJn-abLE9l6dYiGU

takaswie, 2021-09-12:
```
Audio and video is still in slow motion. Same result in VLC. I can see no real difference with this fix.
Packet dump: https://mega.nz/file/figBmCga#NXsEfFFprKNt552jEbjtMrlFXsbrJn-abLE9l6dYiGU
```

Thanks for the test. Hm, I pile another patch into the previous one:

```
oxfw: postpone the cycle to start processing tx packets

The media clock frequency is not stable until enough amount of received
packets are processed. This commit postpones the cycle to start processing
tx packets until 1000 isochronous cycles are past.
```

Would i ask you additional test?

AreYouLoco, 2021-09-12:
`HEAD is now at 51018bd oxfw: postpone the cycle to start processing tx packets`
No real difference.

Dump fixv2: https://mega.nz/file/n3REgBYY#YLA35hobLZnepi3LcErE5df3KWu5lKrePyNmbSIyh5Q

takaswie, 2021-09-17:
@AreYouLoco I'm sorry to be late for reply, but I write additional patch to test. Would you please try with current HEAD (`709535a2879d`) of `topic/fix-oxfw-mackie-regression` branch?

AreYouLoco, 2021-09-21:
I tested I while ago and there seems do be problem since audio is more audible but til not normally to listed.
I Experience XRUNS while its playing and on VLC one seconds breaks i guess separate issue there.
And dump as always: https://mega.nz/file/2roXGY7T#jiwOv0msawAzLzO-C9zoLyggxb5aeqIwJsLwDsC9AMQ

takaswie, 2021-09-21:
> I tested I while ago and there seems do be problem since audio is more audible but til not normally to listed.
> I Experience XRUNS while its playing and on VLC one seconds breaks i guess separate [...]

Thanks for testing. Hm. I attempted to analyze packet dump but the data (`trace_.dat_quiet`) includes no events...
Would I ask you to get the data again?

takaswie, 2021-09-21:
And today I got similar model, Mackie Onyx 820i (OXFW971) in used market, and tested current code. As a result,
current ALSA OXFW driver can successfully handle the model, unlike your Onyx 1640i.

As long as I know, below models have the similar identification information in their configuration ROM:

 * Onyx 820i
 * Onyx 1220i
 * Onyx 1640i

They commonly have two analog output ports and the driver works with two channels of playback.

On the other hand, Onyx 1640i has 16 analog output ports. I think that OXFW 971 has some kind of quirk when
configured to operate for many channels.

Anyway, I'm waiting for the dump data, thanks.

AreYouLoco, 2021-09-21:
Indeed I thought its switched of speakers but there is just no sound. I am using FW channels 1-2 to out. Should try on all 16?!
DUMP: https://mega.nz/file/ijB3VCya#gYkSY0RJSOVkpAXmWTf7QY6v88L7kozcW3MoO8mEgiQ

AreYouLoco, 2021-09-21:
OFTOPPIC:

When you are done with this one I am interested in buying Onyx 820i from you. I hope the sending from wherewere you are is not too expencive.

On September 21, 2021 1:31:54 PM UTC, "坂本 貴史" ***@***.***> wrote:
> And today I got similar model, Mackie Onyx 820i (OXFW971) in used market, and tested current code. As a result,
> current ALSA OXFW driver can successfully handle the model, unlike your Onyx 1640i. [...]

takaswie, 2021-09-22:
> Indeed I thought its switched of speakers but there is just no sound. I am using FW channels 1-2 to out. Should try on all 16?!
> DUMP: https://mega.nz/file/ijB3VCya#gYkSY0RJSOVkpAXmWTf7QY6v88L7kozcW3MoO8mEgiQ
>
> EDIT: I restarted JACK and it seems to produce some sound but still in slow motion.

Thanks, however the content of packet dump indicates that it is retrieved by driver version as of https://github.com/takaswie/snd-firewire-improve/issues/38#issuecomment-917625293.

Would I ask you to reinstall the driver built at current HEAD of `topic/fix-oxfw-mackie-regression` branch? It's now 709535a2879d.

Regards

AreYouLoco, 2021-09-22:
It works. But audio is distorted. Heavily.
DUMP: https://mega.nz/file/L3RQQbTJ#zgMZ4ARryG0-5J9o4ZWE_l864JzMZqIRt7OV7bK-JpU

takaswie, 2021-09-23:
> It works. But audio is distorted. Heavily.
> DUMP: https://mega.nz/file/L3RQQbTJ#zgMZ4ARryG0-5J9o4ZWE_l864JzMZqIRt7OV7bK-JpU

Thanks. We can see apparent improvement in analysis of packet dump:

```
$ trace-cmd report trace.dat_distrorted > list
$ ./parse list 
ch 0:
  5 sec, 513 cycle: 44424 events
  6 sec, 513 cycle: 44048 events
  7 sec, 513 cycle: 43912 events
  0 sec, 513 cycle: 43776 events
  1 sec, 513 cycle: 43608 events
  2 sec, 513 cycle: 43864 events
  3 sec, 513 cycle: 42928 events
  4 sec, 513 cycle: 44352 events
  5 sec, 513 cycle: 43872 events
  6 sec, 513 cycle: 43200 events
  7 sec, 513 cycle: 43424 events
  0 sec, 513 cycle: 43864 events
  1 sec, 513 cycle: 44376 events
  2 sec, 513 cycle: 43520 events
  3 sec, 513 cycle: 41912 events
  4 sec, 513 cycle: 43536 events
  5 sec, 513 cycle: 42760 events
  6 sec, 513 cycle: 43736 events
ch 1:
  5 sec, 255 cycle: 44464 events
  6 sec, 255 cycle: 44032 events
  7 sec, 255 cycle: 43776 events
  0 sec, 255 cycle: 43840 events
  1 sec, 255 cycle: 43608 events
  2 sec, 255 cycle: 43896 events
  3 sec, 255 cycle: 42976 events
  4 sec, 255 cycle: 44288 events
  5 sec, 255 cycle: 43888 events
  6 sec, 255 cycle: 43152 events
  7 sec, 255 cycle: 43520 events
  0 sec, 255 cycle: 43800 events
  1 sec, 255 cycle: 44408 events
  2 sec, 255 cycle: 43480 events
  3 sec, 255 cycle: 41872 events
  4 sec, 255 cycle: 43552 events
  5 sec, 255 cycle: 42752 events
  6 sec, 255 cycle: 43744 events
```

However, we can also see that the media clock frequency is not stable yet. It should be near 48,000
according to current sampling transfer frequency.

Hm. In the attempt, we use 125 msec to start media clock recovery. In next attempt,
let us use double duration, 250 msec. Would I request you for another test with current HEAD of the
remote branch? It's now moved to 544858a14ee0.

takaswie, 2021-09-30:
@AreYouLoco I'm sorry if you are still busy but I'd like you to test the additional patch for the issue.

Thanks.

AreYouLoco, 2021-10-11:
@takaswie I will be on it in two weeks of time. I am sorry I cannot test it earlier. Will get to you as soon as possible after testing.

takaswie, 2021-10-31:
The patch is merged to Linux sound subsystem and waiting for Linux kernel v5.16 prepatch:

 * https://git.kernel.org/pub/scm/linux/kernel/git/tiwai/sound.git/commit/?id=cddcd5472abb7b8a9c37ccbcf0908b79740a01b5

takaswie, 2021-11-15:
Now the patch is queued for linux-stable v5.14 and v5.15:

 * Fixes for 5.14
     * https://git.kernel.org/pub/scm/linux/kernel/git/stable/stable-queue.git/commit/?id=85d6f5e6a4cfcd22a95d46571ea02cf6b67d457d
 * Fixes for 5.15
     * https://git.kernel.org/pub/scm/linux/kernel/git/stable/stable-queue.git/commit/?id=179756d16045dc3812227354f3432c061f4403aa

AreYouLoco, 2021-11-15:
Nice!

takaswie, 2021-11-16:
Linux v5.16-rc1 is out with the patch:

 * https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tag/?h=v5.16-rc1

takaswie, 2021-11-21:
Below stable kernels are out including the fix.

 * Linux 5.14.20
      * https://lore.kernel.org/lkml/1637244207253142@kroah.com/

 * Linux 5.15.3
      * https://lore.kernel.org/lkml/163726325011777@kroah.com/

Let me close the issue, thanks.

closed 2021-11-21 by takaswie