takaswie/linux-firewire-dkms #39: Card detected but not available

epifanio, 2021-09-12:
Hi,

I followed the  installation steps and after the first reboot I was able to see my MOTU UltraLite mk3 Hybrid FW-USB2 in the audio settings .. then suddenly it disappeared and no way I can get it back :(

The card works fine on the same system running windows OS, but I do not use that OS. Just to make sure the fireware pci-e card was working.

```
➜  ~ lspci | grep -i firewire
05:00.0 FireWire (IEEE 1394): Texas Instruments XIO2213A/B/XIO2221 IEEE-1394b OHCI Controller [Cheetah Express] (rev 01)

```
The card seems to be detected properly, see bottom:
```
➜  ~ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: NVidia_1 [HDA NVidia], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: NVidia_1 [HDA NVidia], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: NVidia_1 [HDA NVidia], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: NVidia_1 [HDA NVidia], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: NVidia_1 [HDA NVidia], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: NVidia_1 [HDA NVidia], device 11: HDMI 5 [HDMI 5]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: NVidia_1 [HDA NVidia], device 12: HDMI 6 [HDMI 6]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: NVidia [HDA NVidia], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: NVidia [HDA NVidia], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: NVidia [HDA NVidia], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: NVidia [HDA NVidia], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: NVidia [HDA NVidia], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: NVidia [HDA NVidia], device 11: HDMI 5 [HDMI 5]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: NVidia [HDA NVidia], device 12: HDMI 6 [HDMI 6]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: Generic [HD-Audio Generic], device 0: ALC1220 Analog [ALC1220 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: Generic [HD-Audio Generic], device 1: ALC1220 Digital [ALC1220 Digital]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 4: UltraLiteMk3 [UltraLiteMk3], device 0: FW-MOTU [UltraLiteMk3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

But it does not show up in the audio settings. Neither in pavucontrol, anything I can do to debug the issue?

takaswie, 2021-09-12:
Thanks for the report.

The device, MOTU Ultralite mk3 hybrid is one of my test device, and in my system I can not regenerate such trouble.

I guess that the difference in regard of system might bring the trouble. Would I request you to give information about your system; e.g. the type of distribution, kernel version, your additional configuration, and so on.

epifanio, 2021-09-13:
Sure,
I it is an ubuntu 21.10 running latest kernel available:

`Linux  5.13.0-14-generic #14-Ubuntu SMP Mon Aug 2 12:43:35 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

I did mess around with ffado-svn driver before, with no good results .. I do not know if it has something that generates a conflict with the Alsa driver.
the weird is that for a while I was able to see the mk3 in the audio settings in ubuntu gnome .. 

Which specific information can I provide to help you debugging the issue?

epifanio, 2021-09-13:
OK! Think I found the "non-issue" ... it simply when I switched Input from the gnome audio settings, in the specific .. when I switched from MK3 input the the headset mic... I notice that in pavucontrol  configuration the MK3 device profile goes automatically to OFF .. and then it is no longer available as device for the system.
I didn't know this .. switching it back to "Multichannel Input" gave me a working microphone!

closed 2021-09-13 by epifanio