takaswie/linux-firewire-dkms #41: Firewire MOTU 8-PRE: sound output is VERY distorted

dkdjam, 2021-11-11:
Hey, I just tried the recent kernels and the alsa snd-firewire-motu because I have a couple of MOTU 8-PRE in our rehearsal room. I had good stuff with ffado / jack but I wanted to make it simpler and try your ALSA driver and run without jack.

Setup was a no brainer, but man, the sound coming out is ear-breaking! Sometimes I will hear the real deal, but after a few seconds it degenrates into a loud noise that can't recover. The 1st time I used the driver, it almost made me deaf! 

Another issue: the channel indexing seems completely wrong, output should come out of Main 1 + 2, but it does from 3 + 4. Input channel indexing is also wrong. I plug something on h/w channel 1 but get the stream on s/w input 3. 

When reading your emails / READMEs, I take it that you tried the driver on an 8-PRE, but if you managed to make it work fine, I can tell you it does not at all with mine. FFADO has no issue though, so it must be the ALSA driver. I am not sure how I can help but I am used to linux and troubleshooting things. If you can point me in the right direction, I can find some time to do it. I really hope this can be fixed and improved further because I really want to ditch ffado / jack so that I can switch eventually to pipewire. Thanks for the hard work!

takaswie, 2021-11-12:
Thanks for your report.

> Setup was a no brainer, but man, the sound coming out is ear-breaking! Sometimes I will hear the real deal,
> but after a few seconds it degenrates into a loud noise that can't recover. The 1st time I used the driver,
> it almost made me deaf!

MOTU 8pre FireWire is one of my test devices, and I don't regenerate the issue. I think we have differences
from system environment. Would I request you report about your system environment? For example:

```
 * Linux kernel 5.13
 * Ubuntu 20.04
 * snd-firewire-motu driver in HEAD of master branch (d18fb84e3190), not from kernel package
 * pipewire 0.3.32
 * 48.0 kHz packet streaming
 * optical interfaces are enabled
```

The above is mine.

> Another issue: the channel indexing seems completely wrong, output should come out of
> Main 1 + 2, but it does from 3 + 4. Input channel indexing is also wrong. I plug something
> on h/w channel 1 but get the stream on s/w input 3.

The channel positions comes from packet format decided by MOTU, thus they are not wrong,
unexpectedly to you. The first channel pair is for headphone-1/2, the second channel pair is
for main-1/2 for playback direction. For capture direction,the first channel pair is for mixer-return-1/2,
the second channel pair is for analog-input-1/2.

Thanks.

takaswie, 2021-11-19:
7 days past, but no further information from the reporter. Let me close the issue since it seems to be invalid.

closed 2021-11-19 by takaswie