takaswie/linux-firewire-dkms #43: firewire800 buffer freezes

recallmenot, 2022-05-03:
Hello,
I'm using a FireFace 800 on manjaro and I have an issue.
When I play back audio for an extended period of time, at a random point between 30 seconds and 3 minutes the audio buffer will freeze and be repeated until 5 to 15 seconds after I stop any playback on the computer. Then I can use the interface again.
What I tried that made no difference:
low latency (preempt) or real time kernel
switching to pipewire and pipewire-jack
using pulseaudio, jack or alsa as the source protocol

dmesg --follow gives me no new events when the freeze happens

lsmod | grep snd_fireface gives me:
```
snd_fireface           40960  0
snd_firewire_lib       57344  2 snd_fireface,snd_dice
snd_rawmidi            45056  4 snd_seq_midi,snd_firewire_lib,snd_fireface,snd_dice
snd_hwdep              16384  3 snd_hda_codec,snd_fireface,snd_dice
snd_pcm               159744  7 snd_hda_codec_hdmi,snd_hda_intel,snd_firewire_lib,snd_hda_codec,snd_fireface,snd_hda_core,snd_dice
snd                   118784  15 snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_fireface,snd_timer,snd_pcm,snd_dice,snd_rawmidi
firewire_core          81920  14 snd_firewire_lib,snd_fireface,firewire_ohci,snd_dice
```

readlink -f /sys/bus/firewire/devices/fw1.0/driver/ gives me:
`/sys/bus/firewire/drivers/snd_fireface`

any help solving this would be much appreciated!

takaswie, 2022-05-04:
Thanks for the report.

Would you please investigate the version of Linux kernel installed in your system for my information?

Regards

recallmenot, 2022-05-04:
currently I am on `5.17.1-1-rt16-MANJARO #1 SMP PREEMPT_RT`, latest available Manjaro RT kernel.
I will try 5.4.188 non-RT tomorrow to see whether an old kernel behaves better.
tried increasing pipewire buffer to 4096, no luck.
Very sorry I'm not competent with this. I'll do my best to follow your instructions.
Thank you!

recallmenot, 2022-05-06:
Hello Takashi,
I did some testing yesterday but it was confusing, that's why I didn't write.
I am now on `5.17.1-3-MANJARO #1 SMP PREEMPT`at 96kHz and buffer of 256 samples.
Tested kernel 5.4, no buffer freeze but every 3-5 minutes I got what sounds like bitcrushing.
Switched to current 5.17 kernel, no bitcrushing. Switched back to 5.17-RT, buffer freeze is back.
I then found [rtcqs](https://codeberg.org/rtcqs/rtcqs) and optimized my system for RT kernel but no luck, still I'm getting buffer freezes.
Is there anything else I can do to get RT kernel working?
On current 5.17 non-RT kernel I can load 3 of 4 cores with hyper-threaded mprime (so 6 threads) and playback still doesn't freeze! And much less Xruns than on RT kernel!
Maybe RT kernel reveals a weakness / bug in the firewire / fireface driver that is prevented from occurring in non-RT kernel with preempt?
If I can help you debug this please tell me how, I'd like to help solving this!
Regards

takaswie, 2022-05-07:
> Hello Takashi, I did some testing yesterday but it was confusing, that's why I didn't write. I am now on `5.17.1-3-MANJARO #1 SMP PREEMPT`at 96kHz and buffer of 256 samples. Tested kernel 5.4, no [...]

I think it better to discourage usage of linux-rt unless being kernel developer or kernel enthusiast. See:

- firewire-ohci fails to initialize Texas Instruments XIO2213A/B/XIO2221 based controller on realtime kernels [5.4.91-rt50, 5.10.8-rt24]
    - https://lore.kernel.org/linux-rt-users/YAwPoaUZ1gTD5y+k@hmbx/

recallmenot, 2022-05-10:
ah so I shouldn't use the rt-kernel, well then it's fine.
Thank you for your effort in helping me!
And thank you for creating such wonderful software as snd_firewire!

dreamcat4, 2022-06-20:
@recallmenot could you please let me know which firewire hardware device you were using here? Was it a focusrite product (`snd_dice`) ? Many thanks

recallmenot, 2022-06-20:
hi @dreamcat4 as stated in the OP, it's an RME FireFace 800

dreamcat4, 2022-06-20:
Apologies for waking you up for that! Was in a hurry and totally misread the Fireface as "firewire 800" (which is a different thing entirely, the 1394b standard itself). Very sorry. (but thanks for clarifying)

recallmenot, 2022-06-20:
didn't wake me, no problem, stuff happens