takaswie/linux-firewire-dkms #44: UBSAN: invalid-load in fcp.c load value of 16 is not valid for type '_Bool'

Unaccounted4, 2022-06-09:
Ubuntu 20.04 just got the 5.15 (5.15.0-33) Low Latency kernel in the mainline repo and while booting with snd-firewire-improve installed it throws out that line before plymouth comes up during boot.

Here's the complete error from journalctl:

```
jun 09 14:46:27 AMiGA-3700X unknown: Started bpfilter
jun 09 14:46:27 AMiGA-3700X kernel: firewire_core 0000:05:00.0: created device fw1: GUID 0003db0a00012483, S400
jun 09 14:46:27 AMiGA-3700X kernel: ================================================================================
jun 09 14:46:27 AMiGA-3700X kernel: UBSAN: invalid-load in /var/lib/dkms/alsa-firewire/4.17/build/sound/firewire/fcp.c:362:9
jun 09 14:46:27 AMiGA-3700X kernel: load of value 16 is not a valid value for type '_Bool'
jun 09 14:46:27 AMiGA-3700X kernel: CPU: 2 PID: 424 Comm: irq/24-firewire Tainted: P           OE     5.15.0-33-lowlatency #34~20.04.1-Ubuntu
jun 09 14:46:27 AMiGA-3700X kernel: Hardware name: System manufacturer System Product Name/PRIME X570-P, BIOS 4204 02/24/2022
jun 09 14:46:27 AMiGA-3700X kernel: Call Trace:
jun 09 14:46:27 AMiGA-3700X kernel:  <IRQ>
jun 09 14:46:27 AMiGA-3700X kernel:  dump_stack_lvl+0x4a/0x5f
jun 09 14:46:27 AMiGA-3700X kernel:  dump_stack+0x10/0x12
jun 09 14:46:27 AMiGA-3700X kernel:  ubsan_epilogue+0x9/0x45
jun 09 14:46:27 AMiGA-3700X kernel:  __ubsan_handle_load_invalid_value.cold+0x44/0x49
jun 09 14:46:27 AMiGA-3700X kernel:  fcp_response.cold+0x16/0x23 [snd_firewire_lib]
jun 09 14:46:27 AMiGA-3700X kernel:  fw_core_handle_request+0x24b/0x400 [firewire_core]
jun 09 14:46:27 AMiGA-3700X kernel:  handle_ar_packet+0x1e7/0x210 [firewire_ohci]
jun 09 14:46:27 AMiGA-3700X kernel:  ? handle_ar_packet+0x1e7/0x210 [firewire_ohci]
jun 09 14:46:27 AMiGA-3700X kernel:  ? __enqueue_entity+0x96/0xa0
jun 09 14:46:27 AMiGA-3700X kernel:  ar_context_tasklet+0x144/0x290 [firewire_ohci]
jun 09 14:46:27 AMiGA-3700X kernel:  tasklet_action_common.isra.0+0xe7/0xf0
jun 09 14:46:27 AMiGA-3700X kernel:  ? handle_irq_event_percpu+0x72/0x80
jun 09 14:46:27 AMiGA-3700X kernel:  tasklet_action+0x22/0x30
jun 09 14:46:27 AMiGA-3700X kernel:  __do_softirq+0xda/0x31c
jun 09 14:46:27 AMiGA-3700X kernel:  ? irq_finalize_oneshot.part.0+0xe0/0xe0
jun 09 14:46:27 AMiGA-3700X kernel:  do_softirq+0x66/0x80
jun 09 14:46:27 AMiGA-3700X kernel:  </IRQ>
jun 09 14:46:27 AMiGA-3700X kernel:  <TASK>
jun 09 14:46:27 AMiGA-3700X kernel:  __local_bh_enable_ip+0x62/0x70
jun 09 14:46:27 AMiGA-3700X kernel:  irq_forced_thread_fn+0x7e/0x90
jun 09 14:46:27 AMiGA-3700X kernel:  irq_thread+0xb8/0x170
jun 09 14:46:27 AMiGA-3700X kernel:  ? wake_threads_waitq+0x30/0x30
jun 09 14:46:27 AMiGA-3700X kernel:  ? irq_thread_check_affinity+0xe0/0xe0
jun 09 14:46:27 AMiGA-3700X kernel:  kthread+0x138/0x160
jun 09 14:46:27 AMiGA-3700X kernel:  ? set_kthread_struct+0x40/0x40
jun 09 14:46:27 AMiGA-3700X kernel:  ret_from_fork+0x1f/0x30
jun 09 14:46:27 AMiGA-3700X kernel:  </TASK>
jun 09 14:46:27 AMiGA-3700X kernel: ================================================================================

```

Sound seems to still work fine out of my Apogee Duet (oxfw) after boot completes, so I'm not sure if I should be worried.

takaswie, 2022-06-10:
Hi @Unaccounted4 ,

Thanks for reporting it and sorry for the inconvenience.

I've found the bug March 2022, and posted patch to fixed it.

* ALSA: firewire-lib: fix uninitialized flag for AV/C deferred transaction
   * https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/sound/firewire?id=bf0cd60b7e33

The patch has already been applied to v5.18 kernel. Furthermore it has also been backported to some of stable and longterm kernels under maintained; v4.9, v4.14, v4.19, v5.4, v5.10, v5.15, v5.17.

I wish you got the patched kernels from distributor as soon.

Regards

takaswie, 2022-06-10:
P.S. as you noted, the bug brings neither disadvantage nor corruption to your system. If you are interested, please refer to the patch comment which includes my thought about the bug.

takaswie, 2022-06-10:
Mmm. I did overlook what you report... You reported HEAD of master includes the bug.

Just now I applied the patches and pushed into master. Thanks for your reporting.

Unaccounted4, 2022-06-10:
Can confirm, after pulling latest HEAD and rebuilding dkms it no longer shows the message at boot.

closed 2022-06-10 by Unaccounted4