takaswie/linux-firewire-dkms #48: mic Input no longer avasilable after upgrading to ubuntu 22.10

epifanio, 2022-11-05:
My ultralite mk3 was working fine with this driver, the input mic was detected and usable - after upgrading to ubuntu 22.10 the device is still listed in my audio settings - but no signal is arriving from the microphone. I tested wiring connections and hardware settings on windows - all seems to work properly.

Any clue on how to debug the issue?
Thanks!!!

takaswie, 2022-11-05:
Hi @epifanio,

I'm sorry for your inconvenience. For my information, would I ask you the previous
version of Ubuntu before upgrading?

Regards

epifanio, 2022-11-06:
Hi @takaswie  - I was on on 22.04

takaswie, 2022-11-07:
> I was on Ubuntu 22.04 - I don't know if related but Ubuntu 22.10 moved to a different audio setting - from PulseAudio to PipeWire https://www.omgubuntu.co.uk/2022/05/ubuntu-22-10-makes-pipewire-default

Thanks.

As long as I tested ALSA firewire-motu driver in `linux-modules-extra-5.19.0-23-generic` package (`5.19.0-23.24`), I have no regression between these releases. I think the
change of sound server service from pulseaudio to pipewire brings the apparent
regression to you.

I note that the 1st/2nd channels in PCM frame of capture are for `Stereo Return 1-2 bus`
in vendor's manual, therefore you need to use 3rd/4th channels for XLR microphone inputs.
I guess that pipewire simply deliver the first two channels to your applications, while
pulseaudio multiplexes all of channels to construct a few channels (e.g. stereo) for the
applications.

I think we have the way to configure pipewire connection to use the 2nd pair of channels
instead of the 1st pair, while I'm not so good at pipewire itself. Please contact to pipewire
user for the way.

Cheers.

epifanio, 2022-11-07:
I don't know how to re-configure pipewire - I tried to have a look and got lost - however ... switching back to pulseaudio with:
```
systemctl –user –now enable pulseaudio
```
gave me back a working mk3 audio input

takaswie, 2022-11-07:
> I don't know how to re-configure pipewire - I tried to have a look and got lost - however ... switching back to pulseaudio with:
> 
> ```
> systemctl –user –now enable pulseaudio
> ```
> 
> gave me back a working mk3 audio input

Thanks for the trial. The issue seems to be in the way to use sound server, instead of
regression of ALSA firewire-motu driver. Let me close the issue, thank you.

closed 2022-11-07 by takaswie