takaswie/linux-firewire-dkms #49: Fireface 800 stuck in initialization after sleep

recallmenot, 2023-02-03:
Hello Sakamoto-San,
I want to report an issue: after the computer returns from sleep, my Fireface 800 often does this little "dance" over and over, I don't think it will stop:
[video](https://imgur.com/a/7LVXTou)
I think after sleep the driver continuously tries to re-configure / reset the Fireface but something doesn't work.
I have to turn off the Fireface with its hardware power switch, then turn it back on again and it connects fine again.
This issue has been there in 5.xx kernels also, but it never happened in Windows.
Linux: 6.1.7-1-MANJARO
Firewire card: LSI FW643 PCIe
FFADO is installed but I almost never use / open it. So the driver in use is built-in kernel driver.
Can you please help me in this?

takaswie, 2023-02-04:
Thanks for the report.

> I think after sleep the driver continuously tries to re-configure / reset the Fireface but something doesn't work.

Hm. It is interesting, however I have never tested the device in the case you reported. I'd like to investigate further but
the issue is not so critical. Let me postpone to deal with it.

recallmenot, 2023-02-05:
Thank you Sakamoto-San,
the post-sleep "dance" is not serious, just a mild annoyance.
Do so when you get around to it.
Tell me when I can help you with it.