takaswie/linux-firewire-dkms #50: fireface 802: too much latency with pipewire

francoisromain, 2023-03-01:
Hello Takashi, 
I try to use the fireface 802 on Fedora 37 (Linux 6.1.14) with pipewire (0.3.66). 

Pipewire config `.config/pipewire/pipewire.conf.d/10-clock-quantum.conf` :

```
context.properties = {
        default.clock.quantum       = 256
}
```
The roundtrip latency measured with jack_iodelay:

```
new playback latency: [256, 256]
4753.261 frames     99.026 ms total roundtrip latency
extra loopback latency: 4241 frames
```

This 99,026ms latency makes it unsuable to play instruments. 

* * *
I tried to follow the instructions [here](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/1473#note_1201243)
and modified `/usr/share/wireplumber/main.lua.d/50-alsa-config.lua`:

```
["api.alsa.period-size"]   = 256,
["api.alsa.period-num"]   = 3,
```

But it does not change anything.

How is it possible to fix the latency?

takaswie, 2023-03-02:
Hi,

> But it does not change anything.
> 
> How is it possible to fix the latency?

In my humble opinion, it is preferable to get consultant from WirePlumber developer.

I note that when configured property, node in procfs can report the values during playback, like:

```
$ cat /proc/asound/Fireface802/pcm0p/sub0/hw_params 
access: MMAP_INTERLEAVED
format: S32_LE
subformat: STD
channels: 30
rate: 48000 (48000/1)
period_size: 256
buffer_size: 768
```

francoisromain, 2023-03-02:
Thank you for your answer. 

```
cat /proc/asound/Fireface802/pcm0p/sub0/hw_params
access: MMAP_INTERLEAVED
format: S32_LE
subformat: STD
channels: 30
rate: 48000 (48000/1)
period_size: 1024
buffer_size: 4096
```

The wireplumber config does not seem to be taken in account. 
I am investigating

francoisromain, 2023-03-03:
Dou you have a custom configuration for pipewire / wireplumber?

takaswie, 2023-03-03:
> Dou you have a custom configuration for pipewire / wireplumber?

I'm sorry but I have never customize these services.

francoisromain, 2023-03-03:
`cat /proc/asound/Fireface802/pcm0p/sub0/hw_params` you sent above was done with the fireface 802 and pipewire's default?

takaswie, 2023-03-03:
> `cat /proc/asound/Fireface802/pcm0p/sub0/hw_params` you sent above was done with the fireface 802 and pipewire's default?

Execuse me that it is sample of output in your case. I made the output by hand.

francoisromain, 2023-03-14:
I opened an issue here https://gitlab.freedesktop.org/pipewire/wireplumber/-/issues/433

closed 2023-03-14 by francoisromain