takaswie/linux-firewire-dkms #52: snd-bebob does not recognize Behringer FCA1616

alesandar, 2023-07-21:
I have used an FCA1616 exclusively over USB for more than one year, but just decided to see how it will perform over FireWire, since I relied on `snd-motu` (with MOTU  828mk3) in the past, while using the same laptop and the same FireWire chipset.

<details>
<summary>Hardware/software details about my current setup:</summary>

```yaml
laptop model: ThinkPad W530
integrated firewire controller:  Ricoh Co Ltd R5C832
operating system: Arch Linux
kernel version: 6.3.3.15.realtime2-1-rt
```
</details>

It looks like `snd-bebob` was not loaded automatically after hot-plugging the device and I am sure that it is not blacklisted.
However, `firewire_core` successfully detected a new device and this is the only relevant system log after the hot-plug:
```yaml
firewire_core 0000:02:00.3: phy config: new root=ffc1, gap_count=5
```

What I tried to do next was to load `snd-bebob` manually by running `$ sudo modprobe -vv snd-bebob`, which returns this:
```yaml
modprobe: INFO: custom logging function 0x55de1411caf0 registered
modprobe: INFO: context 0x55de14cb53d0 released
```

I have collected the attribute/value pairs for some of the relevant kernel modules and here is the output for each one of them:

<details>
<summary> $ systool -vm firewire_core</summary>

```yaml
Module = "firewire_core"

  Attributes:
    coresize            = "106496"
    initsize            = "0"
    initstate           = "live"
    refcnt              = "3"
    srcversion          = "67F2C7904F3BB92B859573B"
    taint               = ""
    uevent              = <store method only>

  Sections:
    .altinstr_replacement= "0xffffffffc0936c7c"
    .altinstructions    = "0xffffffffc09371c8"
    .bss                = "0xffffffffc093f000"
    .call_sites         = "0xffffffffc0938864"
    .data               = "0xffffffffc093e600"
    .exit.data          = "0xffffffffc093ec30"
    .exit.text          = "0xffffffffc0936c80"
    .gnu.linkonce.this_module= "0xffffffffc093ec40"
    .ibt_endbr_seal     = "0xffffffffc0939430"
    .init.data          = "0xffffffffc0949000"
    .init.text          = "0xffffffffc0948000"
    .note.Linux         = "0xffffffffc093d4ec"
    .note.gnu.build-id  = "0xffffffffc093d4c8"
    .note.gnu.property  = "0xffffffffc093d488"
    .orc_unwind         = "0xffffffffc0939478"
    .orc_unwind_ip      = "0xffffffffc093bae8"
    .parainstructions   = "0xffffffffc0938518"
    .printk_index       = "0xffffffffc093ec20"
    .retpoline_sites    = "0xffffffffc0938538"
    .return_sites       = "0xffffffffc09385e4"
    .rodata             = "0xffffffffc0937200"
    .rodata.str1.1      = "0xffffffffc0937e54"
    .rodata.str1.8      = "0xffffffffc09381d8"
    .smp_locks          = "0xffffffffc093811c"
    .strtab             = "0xffffffffc094db58"
    .symtab             = "0xffffffffc094a000"
    .text               = "0xffffffffc092d000"
[... 150 lines cut ...]
    .note.Linux         = "0xffffffffc1a3c054"
    .note.gnu.build-id  = "0xffffffffc1a3c030"
    .note.gnu.property  = "0xffffffffc1a3bff0"
    .orc_unwind         = "0xffffffffc1a39dbc"
    .orc_unwind_ip      = "0xffffffffc1a3b23e"
    .printk_index       = "0xffffffffc1a3d460"
    .retpoline_sites    = "0xffffffffc1a39498"
    .return_sites       = "0xffffffffc1a394bc"
    .rodata             = "0xffffffffc1a37000"
    .rodata.cst2        = "0xffffffffc1a39412"
    .rodata.str1.1      = "0xffffffffc1a39040"
    .rodata.str1.8      = "0xffffffffc1a38c08"
    .smp_locks          = "0xffffffffc1a39414"
    .strtab             = "0xffffffffc197ef70"
    .symtab             = "0xffffffffc197c000"
    .text               = "0xffffffffc1a32000"
    __bug_table         = "0xffffffffc1a3d000"
    __mcount_loc        = "0xffffffffc1a38940"
    __param             = "0xffffffffc1a39420"
    __patchable_function_entries= "0xffffffffc1a3d010"

```
</details>

The last thing I did was to compare [my current configuration ROM](https://github.com/takaswie/snd-firewire-improve/files/12127746/FCA1616_config_rom.zip) with [yours](https://github.com/takaswie/am-config-roms/raw/7dce757917e049b8ab20ae6505031bb62a6e9e43/audio_and_music/bebob/behringer-fca1616.img) and all I can say is that they differ from each other.

takaswie, 2023-07-22:
Hi,

Thanks for your report.

> The last thing I did was to compare [my current configuration ROM](https://github.com/takaswie/snd-firewire-improve/files/12127746/FCA1616_config_rom.zip) with [...]

The content of attached configuration ROM is for 1394 OHCI controller. When compiling and install [linux-firewire-utils](https://git.kernel.org/pub/scm/utils/ieee1394/linux-firewire-utils.git/), you can use `config-rom-pretty-printer` program to dump it, like:

```
$ cat FCA1616_config_rom.img | config-rom-pretty-printer 
               ROM header and bus information block
               -----------------------------------------------------------------
1024  04047d6a  bus_info_length 4, crc_length 4, crc 32106
1028  31333934  bus_name "1394"
1032  f000a222  irmc 1, cmc 1, isc 1, bmc 1, pmc 0, cyc_clk_acc 0,
               max_rec 10 (2048), max_rom 2, gen 2, spd 2 (S400)
1036  3c970eff  company_id 3c970e     | 
1040  eb2cf3ff  device_id 1099162252287  | EUI-64 4365974856084026367

               root directory
               -----------------------------------------------------------------
1044  00051b70  directory_length 5, crc 7024
1048  0c0083c0  node capabilities: per IEEE 1394
1052  03001f11  vendor
1056  81000003  --> descriptor leaf at 1068
1060  17023901  model
1064  81000008  --> descriptor leaf at 1096

               descriptor leaf at 1068
               -----------------------------------------------------------------
1068  00064cb7  leaf_length 6, crc 19639
1072  00000000  textual descriptor
1076  00000000  minimal ASCII
1080  4c696e75  "Linu"
1084  78204669  "x Fi"
1088  72657769  "rewi"
1092  72650000  "re"

               descriptor leaf at 1096
               -----------------------------------------------------------------
1096  0003ff1c  leaf_length 3, crc 65308
1100  00000000  textual descriptor
1104  00000000  minimal ASCII
1108  4a756a75  "Juju"
```

I guess that it was retrieved from `/dev/fw0`, while typically `/dev/fw1` is added to system for any node in the bus. Would I ask you to check whether the character device is created or not?

alesandar, 2023-07-22:
>The content of attached configuration ROM is for 1394 OHCI controller. When compiling and install [linux-firewire-utils](https://git.kernel.org/pub/scm/utils/ieee1394/linux-firewire-utils.git/), you can use config-rom-pretty-printer program[1] to dump it...

Nice tip, thank you.

> I guess that it was retrieved from /dev/fw0, while typically /dev/fw1 is added to system for any node in the bus. Would I ask you to check whether the character device is created or not?

That's right, the ROM was retrieved from `/dev/fw0`, because `/dev/fw1` was not present at all.

takaswie, 2023-07-23:
> `/dev/fw1` was not present at all.

It means that the node on your device does not appears in IEEE 1394 bus yet.

As long as I know, the combination of `DIGITAL SELECT` button and powering switches bus mode between USB 2.0 and IEEE 1394. Let you manage to switch the mode before inserting IEEE 1394 plug.

Regards

alesandar, 2023-07-24:
> In my TODO list, FCA1616 does not have [protocol implementation](https://github.com/alsa-project/snd-firewire-ctl-services/blob/master/protocols/bebob/src/behringer.rs) in snd-firewire-ctl-service yet. I guess that the unit can switch source signal for sampling clock, while it is not so urgent.

the digital clock parameter is exposed by ALSA (if connected over USB).