takaswie/linux-firewire-dkms #53: Question...

rockumk, 2023-10-08:
I see Tascam in the read me list. Does this work with the IF/FW or IF/FW mkii cards for Tascam DM4800 and DM3200 mixers?

takaswie, 2023-10-09:
Hi @rockumk ,

> I see Tascam in the read me list. Does this work with the IF/FW or IF/FW mkii cards for Tascam DM4800 and DM3200 mixers?

No one has tested with them yet.

I know that both [IF-FW/DM](https://tascam.jp/int/product/if-fw-dm/top) and [IF-FW/DMMKII](https://tascam.jp/int/product/if-fw-dmmkii/) are application of BridgeCo Enhancement Breakout Box (BeBoB) solution. The protocol of solution is implemented in ALSA bebob driver (`snd-bebob`) driver. The driver has [an entry](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/sound/firewire/bebob/bebob.c#n387) to support `IF-FW/DM`, but I have any seen any report from users.

If you could test with them and report your experience, it is really welcome. The driver have been merged into Linux kernel already, so basically you do not need to install the driver by my repository just for the test.

Regards

rockumk, 2023-10-09:
I don't have a Machine running Linux. I'm active in the DM4800/DM3200 community and was looking for solutions. If I were to test it on a laptop installation, is there a recommended distro for audio? I was actually wondering about trying to create an embedded solution for Dante conversion.

https://www.audinate.com/products/manufacturer-products/dante-embedded-platform

takaswie, 2023-10-11:
> I don't have a machine running Linux. I'm active in the DM4800/DM3200 community and was looking for solutions for all of us as Firewire fades out. If I were to test it on a laptop installation, is there a recommended distro for audio?

I know that there are some similar communities where people interested in such equipment gathers; e.g. [Yamaha 01x](https://www.facebook.com/groups/yamaha01x).

I would be happy if receiving your report to test my driver in any sort of Linux distribution, while it would not be good experience to you, since Linux environment is different from the other type of major operating systems such like Windows. I just focus on implementation of protocol between systems, thus less interests in UI/UX theirselves. Even if my driver handled your device, you would be disappointed about how disadvantaged it is, especially for musical applications.

Nevertheless, if trying it, I would suggest to use [Ubuntu 23.04](https://ubuntu.com/download/desktop), since you can easily get any help from Ubuntu community. Furthermore, I use it as current development environment.

Regards

takaswie, 2024-03-13:
Closed since no reactions during the past few months.

closed 2024-03-13 by takaswie