takaswie/linux-firewire-dkms #57: FF802: system freeze when starting jack

jean-emmanuel, 2024-08-27:
Hi ! Every once in a while I get a system freeze when starting jack (nothing responds, no choice but to hard reboot). It started happening with this driver + the rme FF802 (never happened before with ffado + saffire pro40). Do you have any idea as to how I could track the issue ?

takaswie, 2024-08-28:
Hi,

I think the issue is the one recently fixed by Edmund Raile:
https://lore.kernel.org/linux-sound/20240730195318.869840-1-edmund.raile@protonmail.com/

There are some workarounds:

* Update the kernel running in your system to recent version in which the fixes are appled
* Use PipeWire with JACK API emulation
    * https://gitlab.freedesktop.org/pipewire/pipewire/-/wikis/Config-JACK

jean-emmanuel, 2024-08-28:
Thank you very much, installing it from `topic/backport-to-v6.8/for-next` to make it work on my system (which btw required below patch to build) worked for me. 
```diff
diff --git a/drivers/firewire/sbp2.c b/drivers/firewire/sbp2.c
index 8f7810b2..ca28c9ec 100644
--- a/drivers/firewire/sbp2.c
+++ b/drivers/firewire/sbp2.c
@@ -1532,7 +1532,7 @@ static int sbp2_scsi_slave_configure(struct scsi_device *sdev)
 		sdev->start_stop_pwr_cond = 1;
 
 	if (lu->tgt->workarounds & SBP2_WORKAROUND_128K_MAX_TRANS)
-		blk_queue_max_hw_sectors(sdev->request_queue, 128 * 1024 / 512);
+		queue_max_hw_sectors(sdev->request_queue);
 
 	return 0;
 }
```

No more deadlock / freeze, that a big improvement :). Now it seems the possible root cause is still lying around because it fails to start from time to time but the logs are available (it used to frezze at `ALSA: poll time out [...]`):
```
audio_reservation_init
Acquire audio card Audio0
creating alsa driver ... hw:Fireface802|hw:Fireface802|128|3|48000|0|0|nomon|swmeter|-|32bit
configuring for 48000Hz, period = 128 frames (2.7 ms), buffer = 3 periods
ALSA: final selected sample format for capture: 32bit integer little-endian
ALSA: use 3 periods for capture
ALSA: final selected sample format for playback: 32bit integer little-endian
ALSA: use 3 periods for playback
scan: added port hw:0,0,0 in-hw-0-0-0-Fireface802-MIDI-1
scan: added port hw:0,0,0 out-hw-0-0-0-Fireface802-MIDI-1
scan: opened port hw:0,0,0 in-hw-0-0-0-Fireface802-MIDI-1
scan: opened port hw:0,0,0 out-hw-0-0-0-Fireface802-MIDI-1
ALSA: poll time out, polled for 3998925 usecs, Retrying with a recovery, retry cnt = 1
JackPosixProcessSync::LockedTimedWait error usec = 5000000 err = Connection timed out
Driver is not running
Cannot create new client
Cannot read socket fd = 35 err = Succès
Cannot open qjackctl client
CheckRes error
JackShmReadWritePtr1::~JackShmReadWritePtr1 - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Jack main caught signal 15
ALSA: poll time out, polled for 3998941 usecs, Retrying with a recovery, retry cnt = 2
ALSA: poll time out, polled for 3998950 usecs, Retrying with a recovery, retry cnt = 3
ALSA: poll time out, polled for 3998819 usecs, Retrying with a recovery, retry cnt = 4
ALSA: poll time out, polled for 3998840 usecs, Retrying with a recovery, retry cnt = 5
ALSA: poll time out, polled for 3998982 usecs, Reached max retry cnt = 5, Exiting
JackAudioDriver::ProcessSync: read error, stopping...
scan: deleted port hw:0,0,0 in-hw-0-0-0-Fireface802-MIDI-1
scan: deleted port hw:0,0,0 out-hw-0-0-0-Fireface802-MIDI-1
Released audio card Audio0
audio_reservation_finish
```

takaswie, 2024-08-29:
Let me close this issue since it has been already resolved.

jean-emmanuel, 2024-10-16:
It seems the root cause of the issue is still here, it's not freezing but still failing to start from time to time. `desmg` show the following error when it occurs:
```
snd_fireface fw1.0: invalid position: , pos = 408, buffer size = 384, period size = 128
```

(pos value is not always the same)

jean-emmanuel, 2024-11-05:
@takaswie would you prefer me to open a new ticket for this ?

takaswie, 2024-11-10:
Hi @jean-emmanuel ,

> would you prefer me to open a new ticket for this ?

You address an extra issue against the title of this topic, thus it is natural to file another. However, it is well-known issue as long as I know, therefore it leads no practical action, I think.