takaswie/linux-firewire-dkms #58: DKMS install failed and exit status 2

DrungSturm92, 2024-10-06:
Hello Takashi,
I was trying to install the kernel modules but an unexpected error appeared. I would love to get some help...

/var/lib/dkms/linux-firewire/6.11/build/make.log:

DKMS make.log for linux-firewire-6.11 for kernel 6.8.0-45-generic (x86_64)
dom 06 oct 2024 16:55:44 -03
make W=1 -C /lib/modules/6.8.0-45-generic/build M=/var/lib/dkms/linux-firewire/6.11/build modules
make[1]: se entra en el directorio '/usr/src/linux-headers-6.8.0-45-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-avc.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-trace.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-ci.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-dvb.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-card.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-fe.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-cdev.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-fw.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-rc.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-device.o
  LD [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/target/sbp/sbp_target.o
/var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-device.c:257:18: error: initialization of ‘int (*)(struct device *, struct device_driver *)’ from incompatible pointer type ‘int (*)(struct device *, const struct device_driver *)’ [-Werror=incompatible-pointer-types]
  257 |         .match = fw_unit_match,
      |                  ^~~~~~~~~~~~~
/var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-device.c:257:18: note: (near initialization for ‘fw_bus_type.match’)
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:243: /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-device.o] Error 1
make[3]: *** [scripts/Makefile.build:481: /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire] Error 2
make[3]: *** Se espera a que terminen otras tareas....
make[2]: *** [/usr/src/linux-headers-6.8.0-45-generic/Makefile:1925: /var/lib/dkms/linux-firewire/6.11/build] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-6.8.0-45-generic'
make: *** [Makefile:37: modules] Error 2

takaswie, 2024-10-06:
Hi @DrungSturm92 ,

As the name of remote branch implies, it is for Linux kernel 6.11. You need to revert some commits to backport it to  your 6.8 kernel.

DrungSturm92, 2024-10-07:
Thanks for your reply and your amazing work.

closed 2024-10-07 by DrungSturm92