takaswie/linux-firewire-dkms #61: build on ubuntu noble

epifanio, 2024-11-20:
Hi, following the build instruction, i got the following error:

```
➜  linux-firewire-dkms git:(topic/backport-to-v6.11/main) sudo ln -s $(pwd) /usr/src/linux-firewire-6.11
➜  linux-firewire-dkms git:(topic/backport-to-v6.11/main) sudo dkms install linux-firewire/6.11 --force
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der
Creating symlink /var/lib/dkms/linux-firewire/6.11/source -> /usr/src/linux-firewire-6.11

Building module:
Cleaning build area...
make -j32 KERNELRELEASE=6.8.0-49-generic DKMS_KERNEL_SOURCE_DIR=/lib/modules/6.8.0-49-generic/build modules....(bad exit status: 2)
ERROR (dkms apport): binary package for linux-firewire: 6.11 not found
Error! Bad return status for module build on kernel: 6.8.0-49-generic (x86_64)
Consult /var/lib/dkms/linux-firewire/6.11/build/make.log for more information.

```

I successfully built and used your driver on my previous linux installation [Thanks!!!] (same PC, same audio-card Motu UL MK3 Hybrid) - any help to debug the issue? Is it something to do with my compiler? 

The complete log is:

```
➜  linux-firewire-dkms git:(topic/backport-to-v6.11/main) cat /var/lib/dkms/linux-firewire/6.11/build/make.log
DKMS make.log for linux-firewire-6.11 for kernel 6.8.0-49-generic (x86_64)
Wed Nov 20 10:59:47 PM CET 2024
make W=1 -C /lib/modules/6.8.0-49-generic/build M=/var/lib/dkms/linux-firewire/6.11/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-49-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/target/sbp/sbp_target.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-avc.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-trace.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-ci.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-dvb.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-card.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-fe.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/lib.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-cdev.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/iso-resources.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-fw.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-device.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/packets-buffer.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/media/firewire/firedtv-rc.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/fcp.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-iso.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/cmp.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-topology.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/dice/dice-transaction.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/amdtp-stream.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/oxfw/oxfw-command.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/dice/dice-stream.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-transaction.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/digi00x/amdtp-dot.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/fireworks/fireworks_transaction.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/amdtp-am824.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/fireworks/fireworks_command.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/digi00x/digi00x-stream.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/fireface/ff.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/bebob/bebob_command.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/tascam/tascam-proc.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/motu/motu.o
/var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-device.c:257:18: error: initialization of ‘int (*)(struct device *, struct device_driver *)’ from incompatible pointer type ‘int (*)(struct device *, const struct device_driver *)’ [-Werror=incompatible-pointer-types]
  257 |         .match = fw_unit_match,
      |                  ^~~~~~~~~~~~~
/var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-device.c:257:18: note: (near initialization for ‘fw_bus_type.match’)
/var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-iso.c: In function ‘fw_iso_context_flush_completions’:
/var/lib/dkms/linux-firewire/6.11/build/drivers/firewire/core-iso.c:240:9: error: implicit declaration of function ‘disable_work_sync’ [-Werror=implicit-function-declaration]
[... 75 lines cut ...]
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/bebob/bebob.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/dice/dice-tcelectronic.o
  LD [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/snd-isight.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/dice/dice-alesis.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/dice/dice-extension.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/dice/dice-mytek.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/dice/dice-presonus.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/dice/dice-harman.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/dice/dice-focusrite.o
  CC [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/dice/dice-weiss.o
  LD [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/fireface/snd-fireface.o
make[3]: *** [scripts/Makefile.build:481: /var/lib/dkms/linux-firewire/6.11/build/drivers/firewire] Error 2
make[3]: *** Waiting for unfinished jobs....
  LD [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/digi00x/snd-firewire-digi00x.o
  LD [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/fireworks/snd-fireworks.o
  LD [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/motu/snd-firewire-motu.o
  LD [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/tascam/snd-firewire-tascam.o
  LD [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/oxfw/snd-oxfw.o
  LD [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/dice/snd-dice.o
  LD [M]  /var/lib/dkms/linux-firewire/6.11/build/sound/firewire/bebob/snd-bebob.o
make[2]: *** [/usr/src/linux-headers-6.8.0-49-generic/Makefile:1925: /var/lib/dkms/linux-firewire/6.11/build] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-49-generic'
make: *** [Makefile:37: modules] Error 2
```

epifanio, 2024-11-20:
the foillowing issue seems to be related: https://github.com/NixOS/nixpkgs/issues/342775 which propose a fix here:   https://github.com/medusalix/xone/pull/48/files  
Not sure it is related,

takaswie, 2024-11-21:
As the module version `6.11` expresses, it is buildable just for Linux kernel 6.11. If you would like to build it for the other version of kernels, some remedies should be applied.

Thanks.

closed 2024-11-21 by takaswie

epifanio, 2024-11-21:
Thanks!
I succesfully upgraded my kernel to 6.11 and installed the drivers. I can see my Motu card listed as udio device - but seems is not sending any input signals from the mic. Probably noting to do with the firewire, but more on audio settings on my ditro (pipewire/alsaudio) 

[image]

[image]