takaswie/linux-firewire-dkms #62: Support for linux kernel 6.8

epifanio, 2025-01-30:
Hi,

I was using your keernel on my Motu UL MK3 Hybrid, it was working great :) - but after up[gradfing to latest LTS i have been unable to use it :(
I was wondering if by any chance there will be support for for ubuntu 22.04 LTS (which ship with kernel 5.8) 

Thanks!!!

takaswie, 2025-02-02:
Hi @epifanio ,

Thanks for your report, however this repository is maintained as the bleeding edge for next kernel release (v6.15 or so), so it would not be backported to the old v6.8 kernel.

> I was using your keernel on my Motu UL MK3 Hybrid, it was working great :) - but after up[gradfing to latest LTS i have been unable to use it :(

The support for MOTU UltraLite mk3 (FireWire only) was added to v5.8 kernel, then its Hybrid model was added to v5.14 kernel. Theoretically, the v6.8 kernel also supoprts it, however you saw the regression.

Would I ask your experience when facing the issue?

epifanio, 2025-02-03:
Hi @takaswie , the card is detected and I can see it listed in the audio settings:

[image]

But when I try to select the Audio Input (I have a Mic plugged into the Mic 1/2 input on my ), no audio signal is detected (silent)

[image]

the device is configuread as "Multichannel Duplex"

[image]

Not sure what is wrong or how to debug the issue

epifanio, 2025-03-29:
> Hi [@epifanio](https://github.com/epifanio) ,
>
> Thanks for your report, however this repository is maintained as the bleeding edge for next kernel release (v6.15 or so), so it would not be backported [...]

I updated my system to the latest available kernel 6.11 which seems the one supported by - linux-firewire-dkms - I went ahead and installed the kernel module, but my device now is no-longer listed:

-- alsa-info before installing the kernel module  linux-firewire-dkms: 
    https://alsa-project.org/db/?f=4b6c9e7dcc209f676b7632341dd4766dec732a9e

-- als-info after installing  linux-firewire-dkms:
     https://alsa-project.org/db/?f=8e71f937c46e7051ed4ec9f4eff8932e9058af2a 

Any help is really appreciated, the same hardware works fine on windows, just to exclude hardware issues, and was working with your DKMS ~2y ago when on ubuntu 20/22.04

I also got installed your other SW https://github.com/alsa-project/snd-firewire-ctl-services in the hope to debug the issue . but the card is not found 

Thanks for any help!

epifanio, 2025-03-29:
EDIT, the tree command was not found by alsa-info -- I installed and re-run the utility for a better output:

https://alsa-project.org/db/?f=2379dba21d1325964a53640c71effd68933d7f99

epifanio, 2025-03-29:
Ok, I think this has nothing (or almost) with the kernel module itself, but is all about pipeware vs pulseaudio - I followed some instruction found on reddit [1] - to switch back to pulse audio, and I can now use my MOTU MK3 via firewiwer - I am ble to record using the 2 line-in, it detects the main output but seems it is falling to detect the headphone audio output (at leat on Ardour)

```
sudo apt install pulseaudio pulseaudio-module-bluetooth gstreamer1.0-pulseaudio
systemctl --user stop pipewire.socket pipewire-pulse.socket
systemctl --user disable pipewire.socket pipewire-pulse.socket
systemctl --user mask pipewire.socket pipewire-pulse.socket
systemctl --user enable pulseaudio.service pulseaudio.socket
systemctl --user start pulseaudio.service pulseaudio.socket
```

Thanks for your support.

[1] https://www.reddit.com/r/Ubuntu/comments/1dhttkr/comment/lyw1nl6/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

closed 2025-03-29 by epifanio